iNavFlight/inav #6175: Target F411WSE with more Uarts

Pairan, 2020-10-04:
## Current Behavior
Currently the target has only UART1 and UART2 defined. Additionally Softserial1 and 2 can be used! But somehow this is not enough while there are pins I don't need!

## Desired Behavior
I would like to make the pins something like this:
```c
#define USE_UART3
#define UART3_TX_PIN            PB6 // ### normally defined for I2C1_SCL
#define UART3_RX_PIN            PB7 // ### normally defined for I2C1_SDA
```
But is possible and just that easy? Does the WSE offer 5 UARTS like defined in inav/src/main/target/MATEKF411SE/target.h ?
If so, then a new Target with these new mappings would be awesome and save my day! 

## Who does this impact? Who is this for?
All wing pilots who need another UART for anything like Radar module, camera control or whatever is in need of a fast UART

Pairan, 2020-10-05:
would this be possible like that @MATEKSYS ?

MATEKSYS, 2020-10-06:
not possible.
ominibus target made many guys think TX3/RX3 pins can be used as I2C,  This is limited to certain pins.  It is not a common rule.

Pairan, 2020-10-06:
Okay ... is there anything else I could re-define to make it happen? there are so many things I won't use so it became tempting ;)

closed 2020-10-07 by Pairan