iNavFlight/inav #6176: UPDATE RSSI OSD ELEMENT FROM BETAFLIGHT FOR PROPERLY SCALED CROSSFIRE LQ READINGS (ex 1:100 instead of 99)

LOLDERS, 2020-10-04:
Hi there,

Recently started fling planes with a friend.  We build full RMRC Goblins with INAV and GPS functionality. We have everything dialled in with GPS Hold and RTH working. I fly with Crossfire. As we want to start going for longer range flights, I realized that Inav doesn't have the new Betaflight LQ OSD elements. Could you pull the code and update INAV for us Crossfire users to get LQ displayed the same way as in Betalfight. It is no longer 99 which was a scaling issues. It's now 2 numbers separated with a : 

Thanks!

Liam

Pairan, 2020-10-04:
this should already be done with v2.6 if I remember right ...

OptimusTi, 2020-10-04:
Like Pairan mentioned, this should become available in 2.6. The new stuff shows RFMode : LQ%, RSSI (dBm), VTX Power (mW), and SRN (dB). The GIF below shows a quick antenna null

[image]

DzikuVx, 2020-12-16:
If done, then let's close it

closed 2020-12-16 by DzikuVx