iNavFlight/inav #6189: ESC Telemetry current source do not have a offset current parameter... adding one?

sylphio, 2020-10-06:
## Current Behavior
I am using BLheli_32 esc telemetry as my current sensor source.
Virtual sensor and ADC current sensor both have a "offset" parameter that can be use to take into account the idle current 
usage of the plane. Otherwise, the longer the flight, the more the battery mah counter diverge.

## Desired Behavior 
I suggest to add a offset parameter to the esc telemetry current source or just take into account the existing one.

## Who does this impact? Who is this for?
Anybody using esc telemetry as current sensor source.

sylphio, 2020-10-10:
To help the issue-label-bot a little bit.. this is a feature request.

DzikuVx, 2020-12-11:
blheli_32 has current scaling build in. You can use that to precisely calibrate each of the ESCa

closed 2020-12-11 by DzikuVx