iNavFlight/inav #6191: T-Motor F7 Flight Controller

johnnyo21, 2020-10-06:
I have started looking into the possibility of porting the Matek F722 target file from iNav to work with my t-Motor F7 flight controller.  I have downloaded the source files from the Betaflight Platform for the Matek and the T-Motor Boards.  I have the iNav target for the Matek F722 as a reference.  

Initially I see iNav does not have some of the dependencies for the T-Motor drivers. 

For example (drivers/accgyro/accgyro_mpu6500.c \)  ... is available

while... (drivers/accgyro/accgyro_spi_mpu6000.c \ , drivers/accgyro/accgyro_spi_mpu6500.c \) ... are not.

Is there any documentation somewhere that details how to build a new target, Maybe I need to go back to cleanflight to figure that out.  I am having a hard time finding anything however.

Thanks,

Beardyblair, 2020-10-11:
I was going to port this myself last week but realised the board had no baro. Does your board have a baro? There is space for one but all the ones I have here (10) are not populated. 

I need a baro so stuck with the Matek F722-px.

Vanvaulder, 2021-05-02:
Target f7 hd pro ll

ojmarek, 2021-10-11:
I`m very interested T-MOTOR FC Velox F7 on inav too

closed 2022-01-26 by DzikuVx

Demoskin, 2023-03-04:
Anyone can make a target for t-motors f722 mini. There have a target tmotorf7v2 but not working with my FC. Not working 2 motors and I2C.