iNavFlight/inav #6199: 1 servo does not move

matt454, 2020-10-08:
Hi, I've built a wing using INAV2

It's a SonicModell AR 800 with F411-WING, Caddx Vista, GPS, 2 MG90S servos.

Here is a video: https://youtu.be/smi0j7f6YQ4

One of the servos will only occasionally twitch whereas the other works fine. I have replaced and re-soldered the one that does not work twice and still have the same issue. 

The second issue is that the working servo locks up and only briefly responds to my inputs on the stick.

See the CLI dump here: https://pastebin.com/z52DNBc1

Any help would be great as I've been trying to solve this for a week now.

ChunkyzSoup, 2020-10-08:
I'm no expert. So take this with s grain of sand. But  maybe go to your output tab in inav or betaflight. Enable live mode on servos and play with the weights. See if they move. Also do you need to invert serialrx_inverted = OFF ? Because yours is set to off. I use frsky stuff so i always have yo invert my stuff I use fport so I also have to enable half duplex. Not sure any of this helps. Check and recheck your wiring with marek diagram for your fc. Then go to outputs  and play with the weights and see if they move.  🤷‍♂️ wish I could he more help. But that sums up my knowledge

ChunkyzSoup, 2020-10-08:
Also. Did you go to the mixers tab and load a mixer and save it? That may help.

matt454, 2020-10-09:
Hi, turns out 5 of the 6 servos I bought were faulty. Switching them out for a different brand fixed them. For reference I bought MG90S servos and they didn't work. Replaced with ongrunner KY66.

closed 2020-10-09 by matt454