iNavFlight/inav #6200: Airspeed telemetry for CRSF

MrMap, 2020-10-08:
## Current Behavior

## Desired Behavior

## Suggested Solution

## Who does this impact? Who is this for?

## Additional context

MrMap, 2020-10-25:
I just noticed that Airspeed data gets logged in the Blackbox file - but only the raw values and there are no decoded values.

HenHan85, 2021-06-19:
when using the inav telemetry widget for OpenTX (Horus), the option to choose airspeed sensor (instead of GPS Speed) in the widget setting is not available. Quite sure because there is no telemetry sensor signal coming throug crsf, providing the airspeed sensor signal.
Would be great to have it included for this purpose

https://github.com/iNavFlight/OpenTX-Telemetry-Widget

alexgalstyan, 2021-07-25:
Hello,
Is this issue was resolved ?

closed 2022-02-10 by DzikuVx

pnduffy, 2022-05-09:
I tried getting airspeed telemetry with CRSF protocol, over ExpressLRS, but no airspeed selection is available in the LUA script.  I didn't see if this was closed for a solution or not.  Any comments?

Enggmaug, 2022-09-30:
Hello, 

I'm looking for the very same thing : Pitot data through ExpressLRS CRSF telemetry.

If I understand [this page](https://github.com/iNavFlight/inav/issues/3886) : it should be already possible... But I still don't see how.

uwe-air, 2024-05-29:
Hi same here. PLEASE try to get this done. So i can get an stallwarning working .

pnduffy, 2024-05-29:
I compiled a custom version of INav to get airspeed.  Not sure if it's in the mainline yet.

MrMap, 2024-05-29:
> > Hi same here. PLEASE try to get this done. So i can get an stallwarning working .
> 
> Airspeed shouldn't be used for stall warnings. Stalls can still happen at higher airspeeds. Stalls are caused by separation of airflow from the wing.

It is correct that stall speed (Vs) is not a constant but a function of factors, e.g. weight, configuration and g-load.
However, airspeed is the only speed relevant to flight dynamics, hence including it in the CRSF telemetry protocol is desirable and frankly, surprising that it did not make it´s way in from the beginning.

Though stall speed can vary, using airspeed as a low speed warning is a great idea. This warning indicates that you are flying with low margin to stall speed promt the pilot to take proper action. Typical situation is final approach with an aircraft. On my Avanti S I have my radio read the airspeed repeating every few seconds (using FrSky RX and FrSky series sensor.

I even made a setup in Edge TX adapting the gyro gain to the airspeed, hence optimizing the gyro performance all over the speed range.
Airspeed is not only useful, it´s central.

sensei-hacker, 2024-05-29:
> It is correct that stall speed (Vs) is not a constant but a function of factors, e.g. weight, configuration and g-load. However, airspeed is the only speed relevant to flight dynamics, hence including it in the CRSF telemetry protocol is desirable

 Yes, airspeed is an important parameter if you were trained as / like a full size pilot.

It is ALSO important to understand that you can stall at any speed. Someone who thinks the plane won't stall until a particular speed is also a person whose planes keep inexplicably dropping out of the sky during base turn.

OptimusTi, 2024-05-29:
Airspeed is not in the CRSF protocol. Please reach out to TBS to add it.

MrMap, 2024-05-29:
> Yes, airspeed is an important parameter if you were trained as / like a full size pilot.
>
> It is ALSO important to understand that you can stall at any speed. Someone who thinks the plane won't stall [...]

Yes, good point - but not a reason against having airspeed in the CRSF telemetry protocol.

On a side not, AOA - Angle Of Attack - is a very useful parameter to monitor while staying out of stall trouble.
Unfortunately I have never seen such a sensor for RC use. Maybe to mechanically messy and expensive to translate into RC plane size. But it would be cool.

pnduffy, 2024-05-29:
GA planes have color-coded airspeed indicators showing stall region. It's not an absolute indicator of stall, but during takeoff and landings where the wings are typically level and the angle of attack is not excessive, it's a good indicator. AOA can be detected by the accelerometer if it's calibrated correctly.

sensei-hacker, 2024-05-30:
> Unfortunately I have never seen such a sensor for RC use

I have designed a couple. Both an AoA indicator and a direct stall sensor such as you often see on light aircraft. I can send you the STL if you would like to try it out.

sensei-hacker, 2024-05-30:
> Unfortunately I have never seen such a sensor for RC use.

I've designed a couple. Both AoA and an actual stall sensor. I can send you the STL if you would like to try it out. This style often seem on GA aircraft is probably the simplest and most reliable.

[image]

It's just a toggle switch, which flips up when the stagnation point falls below the set point (meaning the AoA is higher than the selected setting).

For RC use, low friction is required, so I designed it with a cheap Hall sensor for frictionless switching.  The same design turned backwards is an AoA sensor.

uwe-air, 2024-05-30:
Hi i got my ATPL 20 Years ago. We use in the  Aircraft for the  Stallcomputer a mix of :AOA, Speed , flap-setting, Bankangle , Altitude......20millon dots ........etc . Way to complicated to implement this in a hobby modell plane. BUT for what doe we use it in our model?? I dont care if i am high enought to get a stall in a turn . But when i fly nearly straight for landing than the speed i well enogh to use in a model. ( ps the stallspeed dont care about the waight !!!! Its a matter of (non) turbulent airflow on your wing .)

@ sensei-hacker :brilliant don those we had in the fligtshool on the most of the light Aircrafts

james-au, 2024-06-19:
We definitely need airspeed telemetry to set our min/max alarms.  Every scalar telemetry value should have user settable alarms.

There are also "overspeed" conditions (like during a dive) as oscillations can snap a wing quite quickly (been there done that).

On the low end of speed, it's not too uncommon to get distracted with the FPV views and other aspects of flight to forget about a low throttle position.  Especially as digital FPV video does _not_ include audio transmission so the pilot can't even audibly monitor motor throttle as we did with analog FPV.  A "low air speed" alert can save FPV lives!

james-au, 2024-06-19:
> I compiled a custom version of INav to get airspeed. Not sure if it's in the mainline yet.

Excellent.  Can you send me a link to the branch/pull request/diff for this change?