iNavFlight/inav #6202: DJI Remote Control does not work when UART set to DJI VTX

matt454, 2020-10-09:
## Current Behavior
Video feed with OSD in DJI HD goggles works. 
Receiver does not.

If I enable SERIAL RX in ports for UART 1 then DJI Remote will work but the headset will not have an OSD. 

## Steps to Reproduce
Go into ports and set UART one peripherals to DJI FPV HD VTX 
Go into receiver and set to serial and SBUS fast

## Expected behavior
Both DJI receiver and OSD over Video feed work

## Additional context

- FC Board name and vendor: MATEK F411 WING

# version
# INAV/MATEKF411 2.5.2 Aug  4 2020 / 10:25:53 (faaedc7cb)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

Dump here: https://pastebin.com/jbgJiKyA

matt454, 2020-10-09:
[image]
[image]

digitalentity, 2020-10-10:
This is working as intended. DJI HD VTX needs 3 signal lines:

[image]

* UART RX/TX should to connected to a serial port configured as `DJI FPV VTX`
* DJI HDL should be configured to a port configured as `Serial RX`, serial receiver should be set to `SBUS` or `SBUS fast` - aligned with the DJI FPV settings.

closed 2020-10-10 by digitalentity

matt454, 2020-10-10:
Thanks, I followed this and connected the DJI HDL wire to RX, tried, failed, switched to TX on UART 2. On both I set the UART to serial RX and on both the receiver had no connection.