iNavFlight/inav #6203: Static assert fails during build if USE_RX_MSP and USE_MSP_RC_OVERRIDE is defined

aa-calvo, 2020-10-09:
## Current Behavior
`_Static_assert` fails if building with USE_RX_MSP and USE_MSP_RC_OVERRIDE defined

[https://github.com/iNavFlight/inav/blob/master/src/main/cms/cms_menu_osd.c](url)
Line 315

## Steps to Reproduce
1. Try to build any board with USE_RX_MSP and USE_MSP_RC_OVERRIDE defined

## Expected behavior

This static assert should not fail

## Suggested solution(s)

Update the assert to match the expected correct number of menu entries

- FC Board name and vendor: KAKUTEF7
- INAV version string: 2.6

Haceo, 2020-10-11:
same issue, cant seem to figure out how to build now.... any pointers on a temp fix so i can build ;-;

aa-calvo, 2020-10-11:
My suggestion for now would to update the assert.

[https://github.com/iNavFlight/inav/blob/master/src/main/cms/cms_menu_osd.c](url)
Line 315

`_Static_assert(ARRAYLEN(menuOsdElemsEntries) - 2 + 1 == OSD_ITEM_COUNT, "missing OSD elements in CMS");`

to something like this.

`_Static_assert(ARRAYLEN(menuOsdElemsEntries) + 6 == OSD_ITEM_COUNT, "missing OSD elements in CMS");`

EvgenSpinin, 2023-02-04:
также возникает данная проблема F722

 пробовал так
/home/pi/inav/src/main/cms/cms_menu_osd.c:335:1: error: static assertion failed: "missing OSD elements in CMS"
 _Static_assert(ARRAYLEN(menuOsdElemsEntries) + 6  == OSD_ITEM_COUNT, "missing OSD elements in CMS");

EvgenSpinin, 2023-02-05:
> _Static_assert(ARRAYLEN(menuOsdElemsEntries) + 6 == OSD_ITEM_COUNT, "missing OSD elements in CMS");

почему вибрали именно 6 ?

VlatkaMihicORQA, 2024-03-07:
Have someone managed to see into this problem?
I have tried changing the added/substracted number from ARRAYLEN(menuOsdElemsEntries) so the equation is correct but still got the same error.
inav/src/main/cms/cms_menu_osd.c:337:1: error: static assertion failed: "missing OSD elements in CMS"
  337 | _Static_assert(ARRAYLEN(menuOsdElemsEntries) - 2 + 1 == OSD_ITEM_COUNT, "missing OSD elements in CMS");
      | ^~~~~~~~~~~~~~

closed 2024-03-07 by stronnag

VlatkaMihicORQA, 2024-03-07:
@stronnag 
Thank you! I did not see that :)