iNavFlight/inav #6204: Tcmmrc F4129 control firm= matekf405 = MCU: STM32F405

nleo22, 2020-10-10:
i got board Tcmmrc F4129 control firm= matekf405 = MCU: STM32F405   2.5.2 inav
From betaflight=4.1.1 matekf405 everything work find. Bluetooth and rgb working find.
Every since i switch to Inav 2.5.2 STM3f405 bluetooth and rgb led doesn't work anymore please help.

DzikuVx, 2020-10-10:
This board is not oficially supported by INAV. Have you tried to contact the manufacturer?

nleo22, 2020-10-10:
> This board is not oficially supported by INAV. Have you tried to contact the manufacturer?

I did but Noone answer. Even on tcmmrc Facebook 
I'm thinking to replace it. For this HGLRC Zeus F72.
Do you think this one support inav? How do you know? Where to look for...
https://www.amazon.com/gp/aw/d/B08BL5GVR6/ref=ya_aw_od_pi?ie=UTF8&psc=1

closed 2020-12-11 by DzikuVx