iNavFlight/inav #6209: More expo - Ability to set expo above 100

LeoNorw, 2020-10-10:
## Current Behavior
Maximum expo today is 100

## Desired Behavior
Ability to raise expo above 100

## Suggested Solution
Ability to raise expo above 100

## Who does this impact? Who is this for?
All users

## Additional context
More expo would make it possible to make INAV feel more like your favorite settings in any other firmwares so you can fly different models with different firmware without have different behaviors.

MrD-RC, 2020-10-17:
If you need that much expo, your model isn't set up correctly. You probably have too much control surface throws.

LeoNorw, 2020-10-17:
Please read the request before adding irrelevant statements. I don't fly fixed wings, I fly multirotors with high rates.

digitalentity, 2020-10-17:
@LeoNorw nothing in your request implies that it's intended only for multirotors. Please be respectful - this is the best way to have your request considered.

LeoNorw, 2020-10-17:
Do we have to discuss this here? INAV is not able to do the same expo as Betaflight, that has nothing to do with control surface or multirotors for that matter, I just want INAV to be able to do the same or close to the same expo as other firmwares. I do appreciate the effort but please do not try to solve a problem I'm not asking for help solving, I'm asking for a function other firmwares have, so we more easily can jump between firmwares.

digitalentity, 2020-10-17:
INAV base expo formula is the same across INAV, Betaflight and OpenTX radio for pilots flying manual. This makes sense as 70% expo will mean the same thing there.

INAV is not going to have Betaflight's super-expo, mainly because it makes final rate in `deg/s` dependent on the amount of expo applied, this was discussed a few times already.

What can be done as a feature request here:

1. Add full-manual curve configuration so you would be able to create whatever expo curve you want
1. Add support for different style expo (like Betaflight does with KISS, Raceflight etc rates)

LeoNorw, 2020-10-17:
That sounds a bit complicated. I don't know how it works behind the user interface, but wouldn't it be possible to just be able to rais the INAV expo to something like 1.5? For ME that would probably be enough, I don't need to tweak it to be exactly the same, I think that more of what INAV allready is doing would be close enough.

Full manual curve would be a very nice feature, I think others might really love that.

digitalentity, 2020-10-18:
Currently the expo is controlling the blending of linear curve and `pow(x, 3)` curve. More than 100% with existing curve is not possible because it will result in unpredictable behavior. What we could do, is to treat expo >100% as a transition to a more aggressive curve - `pow(x, 5)` or `pow(x, 7)`.

However adding a selector for different types of curves is probably better from usability perspective.

LeoNorw, 2020-10-18:
Thanx for explaining. I hope this is something that could be done, I belive that this would make INAV a stronger alternative to Betaflight.