iNavFlight/inav #6210: Failsafe is not configurable on FlySky I6 equipment with FS -IA6B receiver INAV 2.5.2

Pank172, 2020-10-11:
Failsafe is not configurable on FlySky I6 equipment with FS -IA6B receiver.

When setting up Failsafe in Configurator 2.5.0 with a Matek F405CTR flight controller and firmware 2.5.2 and a FlySky FS-IA6B receiver, Failsafe does not work when the equipment is turned off. When the transmitter is turned off, all the main channels (Roll, Pitch, Yaw) show a channel value of 1500, and the Throtlle and RSSI channels show a value of 1000 and Failsafe does not turn on. When Failsafe is turned on via a toggle switch configured for this function, Failsafe will operate normally. In INAV 2.2.0, in the Failsafe tab of the configurator, there was an item "Valid Pulse Range Settings", where you could set the minimum channel value at which Failsafe was triggered, this is not in the current configurator and I cannot configure this function if the connection with the transmitter is lost.

FC- Matek F405CTR
INAV 2.5.2
receiver-FS-IA6B
transmitter-FlySky FS-I6

breadoven, 2020-10-11:
Valid Pulse Range does seem to have disappeared from Configurator, not sure why but can be set using the CLi and `rx_min_usec`. Setup Failsafe on the Tx for throttle by setting the Throttle min endpoint to the lowest setting (temporary setting only to set failsafe) then set the throttle failsafe below 1000 (-120% = 900). Once the Failsafe is set reset the min endpoint back to the normal value. Set rx_min_usec slightly above the Tx failsafe value.

DzikuVx, 2020-10-12:
For PPM connection radio has to be set for "no pulses" what IIRC this radio can not do or for a special channel combination.
In your case set failsafe on one of the channels to activate FAILSAFE flight mode.

Please bear in mind that PPM is basically obsolete technology no longer used on modern flight controllers and radio receivers

closed 2020-10-12 by DzikuVx