iNavFlight/inav #6211: No servo movement flying wing

Ziov, 2020-10-11:
I've looked online and asked in RC groups but can't get an answer to this issue.

I'm using the Matek f722 Wing FC, have everything wired up and can arm, motor working, gps etc but my servo's won't show any movement. In inav moving sticks and tilting plane shows servo movement but the servo's them selves do not.

Because the servos are being detected and somewhat showing power I'm thinking it could be a settings issue or other.

Could anyone help? Dump file attached.
[1.txt](https://github.com/iNavFlight/inav/files/5361401/1.txt)

DzikuVx, 2020-10-12:
There are no hidden servo settings that are required to make servos work. It's either:
* wrong wiring (wrong outputs)
* servos not powered
* hardware failure

rsilk1949, 2020-10-16:
Are you testing servo movement in manual mode?

MrD-RC, 2020-10-17:
Have you created a mix?

closed 2020-12-11 by DzikuVx