iNavFlight/inav #6212: Cannot arm when in auto launch with throttle > 0%

rsilk1949, 2020-10-11:
Version 2.5.1.
All receiver signals as expected in Configurator & Modes turn blue as expected but ONLY if Throttle is at 0%

Steps:
1 Select angle or manual mode
2 Switch ON for auto
3 Set throttle stick for after launch (e.g. 50%)
4 Switch ON to arm

Also tried the setting to always change to launch mode on arming, but same result.

Here are my launch settings

set nav_fw_launch_accel = 1500
set nav_fw_launch_thr = 1850
set nav_fw_launch_idle_thr = 1150
set nav_fw_launch_motor_delay = 120
set nav_fw_launch_spinup_time = 50
set nav_fw_launch_min_time = 4000
set nav_fw_launch_timeout = 8000
set nav_fw_launch_max_altitude = 5000
set nav_fw_launch_climb_angle = 16

breadoven, 2020-10-11:
Arm then set the throttle stick position as:

https://github.com/iNavFlight/inav/wiki/Modes#nav-launch

rsilk1949, 2020-10-12:
Retested using thus sequence:
Steps:
1 Select angle or manual mode
2 Switch ON for auto launch
3 Set throttle stick for after launch (e.g. 50%)

digitalentity, 2020-10-12:
> I did get the motor to spin at idle_thr (step 4) one time out of 20 attempts.

What is the outcome of the other 19 attempts?

Do you have a beeper connected? Do you hear the arming beep when you flip the arming switch?

rsilk1949, 2020-10-13:
Outcome was the motor never started in idle. Even tried giving it a shake :)
When doing a standard arm without launch switched on and throttle at zero, there are 4 beeps and the throttle is then active.
With the launch switch on prior to arming, there are 4 beeps on arming followed by a continuous sequence of single beeps about 0.25 secs apart.

digitalentity, 2020-10-13:
Can you please record a video (with sound) demonstrating the issue?

rsilk1949, 2020-10-14:
Arming and motor work without launch mode.
See video for case with launch mode

digitalentity, 2020-10-15:
The board is not arming - arming beeps are distinctly different.

The first 3 short beeps usually denote the end of runtime calibration. Odd thing that this happens when you flip the arming switch. The following short beeps are an indication that the arming switch is flipped, but arming is disabled by something - OSD will show the reason.

Try the following:

1. Power the plane, leave it stationary until you hear 3 short beeps marking run-time calibration finish.
2. Flip the launch switch
3. Arm
4. If it doesn't arm - note the OSD message in the goggled (alternatively you can connect the Configurator at this point, enter CLI and execute `status`, which will also show the exhaustive reasons for not arming).

rsilk1949, 2020-10-15:
4. I dont have FPV/OSD :(
Using Configurator, the warning triangle is yellow and says "Arming motor". All pre-arming conditions are green.
CLI will not accept any commands e.g. status, get, exit, etc. I have to disconnect to reset.

digitalentity, 2020-10-15:
Ok, this is getting more and more weird. I'll try to reproduce this.
Please provide a CLI dump of your config so I can try to replicate your issue and figure out what's wrong.

rsilk1949, 2020-10-16:
Resolved. :)
I changed to set Launch mode on Arm switch. 
Armed with throttle at 0.
Continuous sequence of  beeps (assumed launch mode ready signal)
Increased throttle to 100%. Motor started at idle speed "1150"
Simulated launch by shaking.
Throttle  increased to "1900" (take off power)
Throttle then responded normally to stick control.

I aslo made a few minor changes to the launch params which potentially could have made a difference but I doubt it. Hoping to try it at the field tomorrow-- weather permitting.
Tx for your patience and help.

closed 2020-10-16 by rsilk1949