iNavFlight/inav #6214: black box data is incomplete

AndreyOsipkov, 2020-10-11:
## Steps to Reproduce

1. read the log from the black box
2. try to open it in a view, but there is very little data there. Although the file size is large
I don't know if there is a firmware error or the controller itself
---

- FC Board name and vendor: matek f-722mini
- INAV version string: 2.5.2
- INAV-configurator version string: 2.5.0

Beardyblair, 2020-10-11:
What viewer are you using?

AndreyOsipkov, 2020-10-12:
> What viewer are you using?

blackbox-tools 0.4.4
blackbox-log-viewer 3.4.0

theproff777, 2021-02-14:
I am seeing a similar issue.

I am on:
Omnibus F7 Pro
iNav 2.6.0
blackbox-tool 0.4.3
blackbox-log-viewer 3.4.0
INAV-configurator 2.6.0
INAV/OMNIBUSF7V2 2.6.0

I am using OpenLog and even bought a new device thinking there was an issue with the one I had.

gcmcnutt, 2021-06-03:
See if this is related: https://github.com/iNavFlight/inav/issues/7066

stronnag, 2021-06-03:
The log decodes correctly with current `blackbox-decode`

closed 2021-06-03 by stronnag