iNavFlight/inav #6218: OSD : Need feature to make any OSD layout as a Default Layout

sayedidrees, 2020-10-15:
## Current Behaviour
Inav offer to multiple Alternative OSD  layouts but with the lack of channels some people having issue with converting layout to one another, so if the layout is already configured, a feature in inav configuration will make it more easy to make any alternative layout as Default layout.

**More:**
Not everyone  have 10 channel tx or 16 channels, with using 6 channel tx it make channels resources limited, and this factor we cannot use separate channel for alternative layouts, 

*Issue:*
Now when converting the default osd layout to alternative osd layout we are first removing all the elements from the default and then adding elements again  according to out desire and setting them up.

## Suggested Solution
Add a small button with the Drop-down box with text : Make as Default, 
clicking this will swap the default to the alternative layout , and will copy the alternative layout as the Default Layout.

MrD-RC, 2020-10-17:
This is really a configurator issue rather than an iNav issue.

Wouldn't a better option be a feature to copy a layout to another layout? That way, if you want, for example, alt layout 1 on default. You would just copy it to default. If you don't have OSD switching, the copy is fine. Maybe a swap feature would also be useful.

This way, not only can it resolve your issue. But, it also adds functionality to make the creation of alternate OSDs easier. For example, you create an OSD. You want to create a 2nd OSD based on the first OSD, but with more or less data. At the moment, you have to manually recreate the OSD in it's entirety. Having the copy feature would mean that you can just copy the OSD, then modify to your liking.

I know that this is all already possible with CLI, but it would be nice to have it in the UI.

sayedidrees, 2020-10-17:
YEah .. my bad... will  create request their.

sayedidrees, 2020-10-17:
YEah .. my bad... will create feature request their. Copied your Words.

closed 2020-10-17 by sayedidrees