iNavFlight/inav #6222: Altitude Hold For Multirotors using ONLY Rangefinder (Separate Surface Mode)

kidier, 2020-10-17:
## Current Behavior
Current version of INAV is unable to apply AltHold Mode without barometer. Barometer is required for Althold ewen if You have reangefinder, because on difrent altitude difrent sensor works better. And both data are considerate in leveling Multirotor.

## Desired Behavior
Nowdays modern drones are trained to position themselves  using new type of sensors instead of relatively old GPS and Barometer. INAV could be great development platform for such technologies IF there would be a way to Stabilize drone in building or in environment with rapidly changing pressure. Rangefinder and Opticall Flow is perfect solution for such environment. But in inav there is no option to run Poshold and Surface althold without working barometer.

## Suggested Solution
It Would be perfect if anybody could stabilize mid-air drone using ONLY Opflow Sensor and Rangefinder.

For example After detecting only Rangefinder INAV would allow to use Surfacemode, but in difrent conditions.
Surface mode only would be hibryde of althold and ordinary throttle behaviour. 
(My proposition #1)
Throttle 0-24% - drone is on floor/decreasing altitude if in air.
Throttle 25% - mid value for surface mod like 50% in althold.
Throttle 26-49% - MAX climbing speed for surface mode.
Throttle 50% - Current throtle value set/Surface mode not considerate.
Throttle 50-100% - thrust increased proportionally  to throttle values.
(My proposition #2)
Surface mode on it means that drone cannot rise above rangefinder range.
Before takeoff we can type and set max altitude that drone can reach (slightly below rangefinder limit)
Throttle 50% - mid value like in althold mode
After reaching max altitude even if throttle value is above 50% drone stays at previously settled altitude.

## Who does this impact? Who is this for?
It would help A lot young engineering students, software and hardware developers.

digitalentity, 2020-10-17:
One problem with this mode - how would drone behave if the rangefinder doesn't see the surface for some reason (our of range or non-reflective surface)?

Logically - the quad should descend, but without another altitude sensor to back the altitude hold it won't know if it's descending or climbing. The only feasible solution is to cut throttle and drop if rangefinder doesn't return valid data. Anything else would be unsafe.

kidier, 2020-10-17:
> One problem with this mode - how would drone behave if the rangefinder doesn't see the surface for some reason (our of range or non-reflective surface)?
>
> Logically - the quad should descend, but [...]

Of course. Althold using two sources of data is necessary for save flight outdoors, but with one sensor it would be okay for tests.

In situation of range data lost i think it should freeze for 6 seconds and turn on failsafe and it would be up to drone configuration to land safely.
I do agree that it shouldn't be common and open for all feature, because it could lead to some dangerous incidents. But instead i propose Command. Just like "dead reckoning" which allows to use poshold without magnetometer for opflow. Maybe "Single_sensor_altitude = on" would make that future available?

It would help A LOT our research. Please consider such case.

digitalentity, 2020-10-17:
> In situation of range data lost i think it should freeze for 6 seconds and turn on failsafe and it would be up to drone configuration to land safely.

What should it freeze?
Altitude? Not possible without an altitude sensor.
Throttle? What if it was recovering from a descend and had throttle significantly higher than hover - this will result in an uncontrolled climb, hitting the ceiling or flying away.

Revamping the whole position estimation and creating special cases for tackling a total altitude sensor loss is not a valid justification for a one-off and potentially very unsafe edge-case which is easily resolved by adding a cheap barometer to your hardware.

If you need this for your research project - please consider implementing it yourself or reache out to coredev@inavflight.com with a request to get it done by one of the core devs as a paid project.

closed 2020-10-17 by digitalentity

kidier, 2020-10-17:
> Throttle? What if it was recovering from a descend and had throttle significantly higher than hover - this will result in an uncontrolled climb, hitting the ceiling or flying away.

Yes i thought about throttle. And in such case of uncontrolled climb overwright of failsave from rx signal should land him down manually. Such circumstances are only allowed during tests in controlled environment.

Maybe i should star from the beginning. I want to use CrazyBee F4 DX and INAV software as Flight controll system for A.I racing drone. Vision recognition of environment is controlling drone on specified track. And i wanted to use Matek Opflow+Rangefinder to stabilize micro drone.

Speaking of developing such feature for money how much it would cost? That's university project and if you could send back commertial invoice i think it would be possible.

digitalentity, 2020-10-18:
Regarding paid work please reach out to coredev@inavflight.com or to me directly (contacts are in my GitHub profile https://github.com/digitalentity)

FailxIsBack, 2022-08-18:
Hello,

as I'm reading this thread a few years later, I wonder if there are any updates on this topic.
I too would be very interested in having ALTHOLD (actually POSHOLD) without barometer for indoor use together with the Matek Lidar&OptFlow.

What seemed to be the problem here was safety concerns if the rangefinder gives invalid measurements or is out of range. 
I think this could be solved be having a set DESCENDING_THROTTLE = (i.e. 1300).
It might even be possible to use the FAILSAFE LAND mode where the quad is descending at a fixed throttle.
This way, it should be sufficient to simply tell the quad to enter FAILSAFE mode if the rangefinder reports invald. If failsafe is set to LAND the quad will gently descend and land if the THROTTLE value is calibrated correctly.

A more elegant solution would be if the descending state could be stopped once the rangefinder reports validly again so the quad could go into ALTHOLD (POSHOLD) again.