iNavFlight/inav #6223: Bob

boruvka112, 2020-10-17:
Hi guys,I have problem with iNav osd on my Matek F405 wing SE,When i switch on RTH the OSD show AIR,or if i switch to Manual it showr again AIR, I usig firmware 2,5 and Configurator  2,5,0 .The Configurator show switching dots on the mode panel on correct possition  but OSD say just AIR.So i am not sure if i can go to air? Can someone give me advice please?
Thank you Bob

digitalentity, 2020-10-18:
Please refer to the support channels listed here:
https://github.com/iNavFlight/inav#documentation-support-and-learning-resources

boruvka112, 2020-10-18:
Sorry but i don't see there any link to get me to the support.Is somewhwre anodther website where i can get advice?
Thank you Bob

DzikuVx, 2020-10-18:
@boruvka112

closed 2020-10-18 by DzikuVx