iNavFlight/inav #6237: Bigger max loiter radius for airplane current around 100m it to low

srepfler, 2020-10-24:
## Current Behavior
Would not arm with more then 100m radius.

## Desired Behavior
Should be able to loiter in 300m or more radius.

## Suggested Solution
increase max allowed loiter radius or allow to limit be chaneg or set with CLI variable
## Who does this impact? Who is this for?
for automated flying planes, bigger planes, faster planes...
## Additional context
Would be nice to give warnings why it did not armed with bigger radius,
without CLI status did not know why is not arming.

MrD-RC, 2020-10-26:
Personally, I find it really hard to believe that there are planes that need to loiter in a 200m circle (100m radius). Even large motor gliders could happily loiter in half that distance.

Do you have any examples of why you need such a large loiter radius?

I agree about the warnings. You should not be able to enter a value of more than what is accepted.

srepfler, 2020-10-26:
> Personally, I find it really hard to believe that there are planes that need to loiter in a 200m circle (100m radius). Even large motor gliders could happily loiter in half that distance.
>
> Do you [...]

Yes for just loitering it is enough, but people use it as easy alternative to mission like task,
 - cruise around me to check video link quality / antenna position ( if directional range good bad signal) 
- range check
- visually check some position depending on height camera angle, if gimbal even more 
   - search and rescue
   - lost model find

Even 2.4 ghz radio and 5.8 non directional can easy allow 700m circle
It easier to allow bigger loiter than add special modes or advance mission to have programmable radius.

When mission get better, multiple missions, commands relative to position, programmable task, loiter can be less.
For start even to be able to disable check loiter out of limits for people who need more, rest will have default.
Older firmware could loiter in much bigger radius.
Maybe just add warning in configurator ( loiter radius big) and who ignore warning can fly bigger circle.

breadoven, 2020-10-26:
What reason did the CLi status give for it not arming ?

MrD-RC, 2020-10-28:
> Yes for just loitering it is enough, but people use it as easy alternative to mission like task,
> * cruise around me to check video link quality / antenna position ( if directional range good bad [...]

Interesting. I never thought of using loiter like that. I guess because the flight mode is called pos hold, so it indicated just to hold the current position. 

It sounds like for these kinds of tasks it would also be good if it was able to be switched with iNav programming (logic conditions/global functions)? So on a switch, you could have the default radius (for pos hold, as it currently is in iNav), and two different radii for these simple missions.

It also sounds like having pos hold points in missions, where you can set the radius and duration would also be useful too.

srepfler, 2020-10-28:
> What reason did the CLi status give for it not arming ?

Loiter out of radius. ( Something like that - status is great, but without status no clue)

breadoven, 2020-10-28:
There is a problem where Configurator lets you enter invalid values through the settings tabs and saves them happily to the FC, least it does for Loiter Radius. However, if you try to enter the value through the CLi it won't let you so this needs fixing, ... Configurator issue.

I guess it's worth checking the CLi status after changing settings to see if there are any errors like this before you head off to fly.

stronnag, 2020-10-28:
2.6 configurator validates this correctly. Values > 10000 are set to 10000.

breadoven, 2020-10-28:
I was using Configurator 2.6 Version 0.42.2.0. Is there a newer version ?

stronnag, 2020-10-28:
tested with git commit 6607343e on 2020-10-26 via http://seyrsnys.myzen.co.uk/inav-configurator-next/

breadoven, 2020-10-28:
If you change focus to something else the invalid value does change to the max/min limit. However, It seems there is a bug related to changing the value and then clicking Save and Reboot without changing focus from the field you changed. So enter 15000 in the Loiter Radius field and click Save and Reboot with the cursor still in the Loiter Radius field and it saves 15000 to the FC, which then won't Arm. Just realised on reboot it shows a red cross for "Settings Validated" in the Setup tab which shows settings are invalid although you'd still need to use CLi status to understand why.

MrD-RC, 2020-11-03:
I can confirm @breadoven 's findings. The data validity isn't checked until the field loses focus. So any field could have incorrect data filled and submitted.

@srepfler having a look a little deeper. It wouldn't be good to set the loiter radius so high by default. I think the 100 m radius is correct. This is because other navigation modes also use the loiter radius, such as RTH and waypoints. 

However, I think that setting larger radii via programming would be ok. This is because it would be separate from the nav functions and only override pos hold when the switch is set to change the radius. It would also allow you to have multiple alternate radii for testing. As a safety feature, I think that the minimum value should be no less than the standard loiter radius (can be constrained in code). But it could allow for a much larger radius, for example 1000 m.

mirov, 2021-01-18:
Today I came across the 100m limit as well.  I'd tried to set it to 200m so that with the new 2.6 Safehome settings I could have the plane RTH and loiter well away from the line of tall trees that are at my back when I launch.  A little farther back is a freeway, so we really don't want to loiter anywhere near the launch point.  I'm flying a fixed wing motor glider and a 100m radius is actually tighter and steeper bank angle (and faster) than I'd like.  Here's a screenshot from the FPV cam today showing the field geometry (Baylands Park, the same location as the OSD setup screen background image btw):
[image]

The red line is the freeway, just to the right is the line of tall trees, and we tend to launch in the shadow of the trees.  I tried to set my Safehome point at 280m away (and ran into its 200m limit) so that with a 200m loiter radius the plane would have an arc right through the middle of the field (or close to it).

For me this is also an easy way to have a canned Nav mission that I can use to test a new battery pack, radio link, VTX settings, etc.

MrD-RC, 2021-03-03:
I have started work on this. It will not increase the standard loiter radius settings, as I believe the maximum is a decent, safe distance for most pilots. They will however allow you to increase the loiter radius beyond that, with mission tasks as mentioned above in mind. This would need to be done with iNav programming. It will allow you to have different loiter radii on a switch. The base code is there, but I need to get out and test it.

https://github.com/MrD-RC/inav/tree/set-loiter-radius-with-programming
https://github.com/iNavFlight/inav-configurator/compare/master...MrD-RC:set-loiter-radius-with-programming

[image]
This is an example of the programming I'll be using to test. The GVAR setting (lines 4 & 5) are so that I'll also be able to see the intended loiter radius on my OSD, in metres.

mirov, 2021-03-03:
> iNavFlight/inav-configurator@
> master...MrD-RC:set-loiter-radius-with-programming

I like the idea of doing this through the programming interface, I could, I
suppose, even use one of the knobs on my transmitter to set the radius.  Is
it dynamic (ie: if I turn my hypothetical knob will it change) ?

-Russ

MrD-RC, 2021-03-03:
@mirov in theory yes. But you’d need to do some computation on the pot values from a 1000 to 2000 range to an X to 50000 range. Or maybe use a curve to have specific steps on the pot?

But if you’re going free range, I’d definitely have something on screen to display the loiter radius too. Maybe its worth adding a specific loiter radius OSD element?

mirov, 2021-03-03:
I sure don't mind doing a little math to scale and probably limit the
target radius.

BTW, between the start of this thread and today I've been flying full WP
missions.  Twice now I've taken off without transferring the mission from
EEROM to RAM and been momentarily puzzled when my plane flips into RTH mode
(because it doesn't have a mission loaded).   Some OSD indication of
mission readiness would  be great.  I wonder if it would be too extreme to
have an option where the FC won't arm if there's a mission available but it
isn't loaded...  Or perhaps to have an option to auto-load into RAM a
mission if it is available.  That seems pretty benign...

-Russ

avsaase, 2021-03-04:
@mirov see #6612

MrD-RC, 2021-03-04:
@mirov Got it working with a pot. I'm uploading a video to show it in action on the ground.
[image]

PR #6800 (2021-04-09): Increase loiter max radius to 300m
closes #6237
closed 2021-04-09 by DzikuVx