iNavFlight/inav #6239: RTH on failsafe overshoots home location

jcameron, 2020-10-26:
## Current Behavior
Upon RX loss, my 7" quad with INAV 2.5.1 flies towards the home point, but instead of landing it keeps going in the same direction until the RX link is regained. As you can see in https://www.youtube.com/watch?v=vAMIJ_Nc84I , the distance to home decreases to zero, then starts to increase with no change in direction. The home direction arrow was correct throughout, and the quad was several hundred meters away before I regained control.

Once I had control back, I triggered a manual RTH using a switch and it immediately returned home and landed at the takeoff location. During this entire event there was very little wind, and I was getting a consistent 12 GPS sats. The status message in the OSD was consistently showing "RC RC LINK LOST" and "EN ROUTE TO HOME". 

I also noticed that during this failsafe RTH, it was returning to home at only 10 kph, while during a manually triggered RTH it reached 40 kph (as configured). Also, the same quad with the same settings has successfully returned to home and landed in manually triggered RTH mode many times in the past.

## Steps to Reproduce
1. Take off in GPS hold mode and head out a few hundred meters
2. Power off transmitter
Note that I haven't dared to intentionally re-trigger this behavior!

## Expected behavior
Upon failsafe, quad should return to home, hover over the home point, and then land. 

## Suggested solution(s)
RTH on failsafe should behave exactly the same as manually activated RTH.

## Additional context
Full CLI dump : https://www.webmin.com/full_cli_dump.txt

The RC link on this quad uses Dragonlink, with the transmitter module connected to a Spektrum DX6. It's also unclear why the Dragonlink connection dropped to trigger this, but I was able to regain control by turning the Dragonlink transmitter off and on again. Regardless of that, INAV did correctly detect this as an RX loss and should have returned home. 

---

- FC Board name and vendor:
Not sure of the board name, it came included with this quad from Banggood : 
Skystars 2019 G730L GPS INAV 300mm F4 8K OSD FPV Racing Drone PNP BNF

- INAV version string:
# INAV/MATEKF405 2.5.1 Jun 16 2020 / 12:41:44 (098a574b3)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

stronnag, 2020-10-26:
Please provide a blackbox log.

stronnag, 2020-10-26:
`set nav_extra_arming_safety = OFF` == all bets are off 
Please also reproduce with `set nav_extra_arming_safety = ON`

jcameron, 2020-10-26:
Sorry, I don't have a blackbox log - even though an SD card was inserted and logging enabled in the configurator, nothing was saved to the card.

I will turn nav_extra_arming_safety to ON because it's a good idea regardless, but from reading the docs this option only effects the required GPS lock at arming time. I could understand if arming too early caused the home point to not be recorded, but based on what I saw in the OSD it was recorded - just not correctly navigated to in failsafe RTH.

breadoven, 2020-10-27:
Well the closest it got to the home position was 3m according to the OSD, and your `nav_wp_radius `is set to 1m (default) so it seems it initially missed Home which didn't trigger Hover and Land, hence stuck on "EN ROUTE TO HOME". Should have navigated back having missed, but for some reason decided not to ... odd.

jcameron, 2020-10-27:
On this same quad, I've seen manually triggered RTH miss the home point, but then circle around until it's above the target and land.

Is the code path for failsafe RTH the same as manual RTH? Because during this incident, I also noticed that was returning to home much slower on failsafe than manual.

jcameron, 2020-10-31:
Yeah, I'll see if I can reproduce this with logging enabled.

Is some action required to begin logging, or does it just start as soon as the quad is armed?

breadoven, 2020-10-31:
https://github.com/iNavFlight/inav/blob/master/docs/Blackbox.md

closed 2021-03-02 by DzikuVx