iNavFlight/inav #6241: Custom OSD message(s) on logic condition

avsaase, 2020-10-28:
## Current Behavior

#6195 implements the display of gvars on the osd. 

## Desired Behavior

It would be nice to also be able to display custom messages on the osd based on logic conditions. For example on a fixed wing aircraft, if you have flaps on channel 7, it could display "flaps middle" when channel 7 is in the middle position and "flaps down" when channel 7 is low. Another application is to have a "too far from home" message on the osd when you fly more than x meters away from the home position.

## Suggested Solution

Create a custom text message operation in the programming framework. The osd tab then needs a section where these messages can be enabled and put on the osd. Alternatively there could be a single custom message that can be used by multiple logic conditions. 

## Who does this impact? Who is this for?

The applications of such a feature are virtually endless.

MrD-RC, 2020-10-28:
This sounds like a cool idea. The "Too far from home" message would be much clearer than the blinking home distance with the alert.

I guess these could be displayed in the system messages area.

avsaase, 2020-10-29:
> I guess these could be displayed in the system messages area.

Yes this would probably be easiest to implement and also to fit on the osd. Separate shorter (less osd character) messages that you can place around the osd wherever you want would allow for more flexible applications, though.

ShawnzAZ, 2020-10-30:
Some other use cases that would be nice for this:

Gear up / down
door open (bomb bay, rear door / whatever)
spoilers

Pretty much things you can see flying LOS, but cannot usually visually verify when flying FPV.

Beardyblair, 2020-11-22:
totally agree, great idea. Ive suggested it a few times in Pawels live chats but its not be answered yet.

avsaase, 2020-12-04:
Humbly paging @DzikuVx

mnwachukwu, 2021-04-20:
Found this issue looking for if this is possible... I guess as of right now, it is not.

I'm going to be taking a crack at this.

b14ckyy, 2021-05-10:
Dang, I just had the exact same idea when i saw a video of a FPV plane with landing gear. And found this ticket. 
I really think that would be a great benefit. 

This could be nice to check logic confition functionality too, like my automatic LED control on the Talon GT that tells me when LEDs are switched on or off automatically based on altitude and distance. Or a warning to lower the landing gear when you are close to home and low or to retract landing gear if you are high and further away. 

Really endless possibilities.

b14ckyy, 2024-03-20:
Implemented with https://github.com/iNavFlight/inav/pull/9508

closed 2024-03-20 by b14ckyy