iNavFlight/inav #6244: MS4525 Differential Pressure (Airspeed) Sensor Bug

ethbond, 2020-10-29:
I believe there is a bug with the driver or error detection for the MS4525 although I am not sure what.

I have two ms4525, one pixhawk and one cuav, both work fine when plugged into an Arduino but when plugged into either my kakute f7 or matek f411 boards INAV reports an airspeed sensor failure. I have eliminated the possibility of a faulty i2c connection by plugging in compass in parallel and on its own, which works fine. Going back a couple of versions of INAV the problem still persists, and changing the i2c frequency has no effect.

avsaase, 2021-04-23:
I have the Matek digital airspeed sensor, which uses the same sensor, and it works fine. However on my setup it doesn't get power from usb and if I plug in battery after connecting it gives me the error you described. If I first connect connect the lipo and then usb it works as expected.

b14ckyy, 2024-03-20:
Is this still an issue?

closed 2024-08-27 by DzikuVx