iNavFlight/inav #6245: ZEEZ F7 + VTX power on

root1911, 2020-10-30:
Hello everyone, does anybody use INAV with ZEEZ F7? I have a problem with VTX activation. In Betaflight it is activated with User1 mode but in inav this option is not working. On ZEEZ VTX is disabled by default.
Is there an option to set this? if yes, please give instructions.

---
- FC Board name and vendor: Zeez F7
- INAV version string: faaedc7cb (INAV/ZEEZF7 2.5.2)

NameOfTheDragon, 2021-01-09:
> Hello everyone, does anybody use INAV with ZEEZ F7? I have a problem with VTX activation. In Betaflight it is activated with User1 mode but in inav this option is not working. On ZEEZ VTX is disabled by default.
> Is there an option to set this? if yes, please give instructions.

Did you ever resolve this?? I'm just having the same issue, VTX is disabled. I have 5 volts at the VTX 5v/Vbatt jumper but nothing at the V_TX output pin. There's an LED labelled "VTX ON" right next to the jumper and that's off.

Zeez doesn't seem to have made any of their design public so it is very difficult to troubleshoot this sort of thing.

NameOfTheDragon, 2021-01-10:
OK I got mine working. The target needs to have PINIO features enabled. I've submitted a pull request with this (PR #6492 ). If you are not starting from defaults, you may also have to do some manual config in CLI:

<pre>
# set pinio_box1 = 40
pinio_box1 set to 40
</pre>

The above command associates `PINIO1` (which is mapped to the VTX power switch, pin PB11) with `USER1` in the modes screen.

DzikuVx, 2021-03-02:
fixed in #6492

closed 2021-03-02 by DzikuVx