iNavFlight/inav #6249: F765 Ports settings not saving

bencoton, 2020-10-31:
## Current Behavior
Matek F765 new setup - all on board sensors work. When trying to add a TBS GPS module I can not get the ports tab to save the settings.

Tried multiple ports  multiple connections and just can not get the ports settings changes to save. Every time it reboots, the original settings are present.

I have tested the GPS using a USB to serial converter with my laptop and there are no problems (U-Center)
[image]

## Expected behavior
I would expect any changes to the ports to save regardless of the hardware connected.

CLI Dump:
https://pastebin.com/0zkv3vTk

- FC Board name and vendor:
Matek F765
- INAV version string:
GCC -9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

stronnag, 2020-10-31:
Enabling `feature GPS` might just help.

DzikuVx, 2020-10-31:
Also, it's probablyu you try to have 2 functions on one port at the same time. LIke MSP and GPS. This will not work. One porta - 1 function

closed 2020-10-31 by DzikuVx

bencoton, 2020-11-01:
Ok had a think about it and tried selecting the sensor (GPS) and nothing else. It now works. I assumed you needed to turn on either the MSP or the Serial RX slider. 
[image]