iNavFlight/inav #6254: Stick down, Throttle up

sebakrys, 2020-11-02:
Hi,
Last week i experienced strange situation with my quad on iNav flying.
I was testing the althold feature while flying indoor. Then suddenly drone started gaining altitude, without any reason(as far as i know). I countered moving the Throttle stick down. Drone reacted to this, but when i stopped lowering the throttle he started again increasing the throttle until max value and then it hit the ceiling. Everything i said you can see in blackbox screenshot. I showed blackbox to my friends who have more experience with drones and they said that they also see no reason why this situation happened. 
[image]
You can see, i decreased throttle to zero and quad kepd increasing the motors speed.
I'm asking you to help me with this by telling me the reason for this, or maybe some clues why this happened and how to not allow this to happen next time. In additioon i want to say that this happend only once, and never again (during week).
I also attach blackbox files for you to analyze

[LOG00237.TXT](https://github.com/iNavFlight/inav/files/5472523/LOG00237.TXT)

DzikuVx, 2020-11-02:
[image]

The log indicates that:
1. at 10.7s there was something that caused baro to drop readouts from 0,4m to -3,6m in less than 1 second
1. INAV started to compensate by increasing the throttle
1. baro readots go back to normal but at the same time, amount of vibration raises that most probably started fooling around with altitude estimation
1. From this point, due to big baro spike and vibrations altitde estimator record altitude drop, not raise
1. there is no GPS to help with altitude estimation

Suggestions:
1. Secure baro with a open cell foam to mitigate wind gusts 
1. Investigate and eliminate excessive vibrations on high throttle
1. Add a GPS

sebakrys, 2020-11-02:
@DzikuVx  Thank you for your response and suggestions, I also forgot to mention that on board i have **DPS310** barometer instead of BMP280, and custom iNav version from MATEK for this particular barometer(inav_2.5.1).

closed 2020-11-26 by DzikuVx