iNavFlight/inav #6264: HGLRC Zeus F722 mini Wing/Plane Motor Outputs unreachable

fchansa, 2020-11-04:
## Current Behavior
With iNav 2.5.2 and the Wing Mixer the motor outputs are @ S7/8.
Unlike his big brother HGLRC Zeus F722, the MINI version has no S5/6/7/8 Pads,
only a 4in1 AiO Connector with S1/2/3/4.

## Steps to Reproduce
1.Flash iNav
2.Load the mixer (wing for example)
3.Look at the outputs (motors are S7/8)

## Expected behavior
motor output(s) is S3 or S4

## Suggested solution(s)
Change motor output to something reachable :-)

- FC Board name and vendor: HGLRC Zeus F722 mini
- INAV version string: 2.5.2

DzikuVx, 2020-11-05:
Will not happen as S1-S4 share the same Timer setup and thus have to be used for the same function

fchansa, 2020-11-05:
Ok - this means the HGLRC Zeus F722 mini is unusable for a wing/plane?
WTF - why can i use other boards with only 4 PWM outputs (NOX, Airbot Nano OMNIBUS F4 V6, ...)
and this board not? What's the difference?

DzikuVx, 2020-11-05:
Not every pin can be used for every function. Hardware limitation. Short answer:

hglrc - S1-S4 the same timer
[image]

Nox - different timers
[image]

Long answer: https://www.youtube.com/watch?v=v4R-pnO4srU

fchansa, 2020-11-05:
I heard about the problem with the timers - thanks for the explanation Pavel. On the other hand, this also means that the HGLRC Zeus F722 MINI is NOT suitable for planes/wings? So i have to buy a new FC :-) ?

DzikuVx, 2020-11-05:
You can connect motor to a servo output and experiment with it

closed 2020-11-05 by DzikuVx

fchansa, 2020-11-05:
But will the flightmodes with automatic throttle handling (autolaunch etc.) work with this crazy configuration?
How will iNav know that there ist the motor not a servo (for manual, acro it could work.)

DIY420, 2021-03-21:
I'm struggeling at the same issue actually. I want to use the mini because  of space restriction in my Amokka fuselage.
Earlier I pimped the OmnibusF4pro target for 6 servos + 2 motors in INAV221, the only thing to be aware was the fact that using the same timer for both, motor and servo, has the consequence that both are 50Hz or both are 400Hz, but then its possible. But it's a quite huge effort for me to recompile the complete INAV stuff via Docker and so on, because this is not my world...

In my opinion it should work to recompile with new target.c timer channel assignment, instead of actually using
const timerHardware_t timerHardware[] = {
    DEF_TIM(TIM3, CH1, PB4,   TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S1   UP1-2   D(1, 4, 5)
    DEF_TIM(TIM3, CH2, PB5,   TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S2   UP1-2   D(1, 5, 5)
    DEF_TIM(TIM3, CH3, PB0,   TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S3   UP1-2   D(1, 7, 5)
    DEF_TIM(TIM3, CH4, PB1,   TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S4   UP1-2   D(1, 2, 5)
    DEF_TIM(TIM2, CH1, PA15,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S5   UP1-7   D(1, 5, 3) - clash with S2
    DEF_TIM(TIM2, CH2, PB3,   TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S6   UP1-7   D(1, 6, 3)
    DEF_TIM(TIM4, CH1, PB6,   TIM_USE_MC_SERVO | TIM_USE_FW_MOTOR, 0, 0),   // S7   UP1-6   D(1, 0, 2)
    DEF_TIM(TIM4, CH2, PB7,   TIM_USE_MC_SERVO | TIM_USE_FW_MOTOR, 0, 0),   // S8   UP1-6   D(1, 3, 2)

the alternative solution would be to use:
const timerHardware_t timerHardware[] = {
    DEF_TIM(TIM3, CH1, PB4,   TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR, 0, 0),   // S1   UP1-2   D(1, 4, 5)
    DEF_TIM(TIM3, CH2, PB5,   TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR, 0, 0),   // S2   UP1-2   D(1, 5, 5)
    DEF_TIM(TIM3, CH3, PB0,   TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S3   UP1-2   D(1, 7, 5)
    DEF_TIM(TIM3, CH4, PB1,   TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S4   UP1-2   D(1, 2, 5)
    DEF_TIM(TIM2, CH1, PA15,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S5   UP1-7   D(1, 5, 3) - clash with S2
    DEF_TIM(TIM2, CH2, PB3,   TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S6   UP1-7   D(1, 6, 3)
    DEF_TIM(TIM4, CH1, PB6,   TIM_USE_MC_SERVO | TIM_USE_FW_SERVO, 0, 0),   // S7   UP1-6   D(1, 0, 2)
    DEF_TIM(TIM4, CH2, PB7,   TIM_USE_MC_SERVO | TIM_USE_FW_SERVO, 0, 0),   // S8   UP1-6   D(1, 3, 2)