iNavFlight/inav #6265: Board Alignment Adjustment Issue, v2.5.1

Mateyhv, 2020-11-05:
There is an issue with the adjustment of the Board Alignment on Roll and Pitch through the Adjustment tab.
Board is set at -91º Yaw. Adjustment is done via the Radio, one chanel drives the adjustment mode, the other on a three position instant switch does the increment/decrement of the selected value.

- When Pitch Adjustment is selected AP starts blinking on OSD however the value changed in any direction is AR in 0.5º increments. Simultaneously AP is changed in 0.1º  fractions but only on decrementing switch until it reaches 0.0º, then it is not affected anymore. If inital AP value was positive its decrementing. If initial AP value was negative its incrementing, in any case after reaching 0.0º doesn't change anymore.

- When Roll Adjustment is selected AR starts blinkin on OSD however the value changed in any direction is AP in 0,5º increments. Simultaneously AR is changed in 0.1º  fractions but only on decrementing switch until it reaches 0.0º, then it is not affected anymore.  If inital AP value was positive its decrementing. If initial AP value was negative its incrementing, in any case after reaching 0.0º doesn't change anymore.

Since I haven't seen this issue before I believe it is ported to v2.6

Thanks

digitalentity, 2020-11-05:
Configured adjustment values are relative to the board. Since your board is rotated 90 deg, when you want to adjust pitch (relative to the aircraft), board adjusts roll (because relative to the board it's roll).

Since your align/board yaw is not exact multiple of 90 deg - FC adjusts pitch and roll simultaneously, in the smallest increments possible (0.1 deg).

This is actually an edge case. Align yaw is intended to change the orientation of the board to conveniently mount the board. Usually it's a multiple of 45 deg. What's your use case that requires exactly 91 deg and 90 is incorrect?

Mateyhv, 2020-11-05:
Thanks for the quick answer Konstantin!

> Configured adjustment values are relative to the board. Since your board is rotated 90 deg, when you want to adjust pitch (relative to the aircraft), board adjusts roll (because relative to the [...]

That is what I supposed but was missing the confirmation.

> This is actually an edge case. Align yaw is intended to change the orientation of the board to conveniently mount the board. Usually it's a multiple of 45 deg. What's your use case that requires exactly 91 deg and 90 is incorrect?

The FC is mounted on a double sided tape. Because of some tensión the board has rotated slightly so I compensated through Yaw until the "brick"  and the model pitch up and down perfetly sinchronized. Knowing that Yaw adjustment is intended for 45º increments I will put it back to -90º and wont mess with it. Closing the issue.

Thanks
Matey

closed 2020-11-05 by Mateyhv

digitalentity, 2020-11-06:
To circle back - 1 deg error shouldn't matter much. I usually don't bother if misalignment is barely noticeable  eye without using measurement tools.