iNavFlight/inav #6272: Autothrottle broken in INAV 2.6 Alfa 2 - Throttle inverted in turns

b14ckyy, 2020-11-07:
## Current Behavior
In 3D CRUISE, POSHOLD and RTH the throttle is massively reduced when autothrottle is active. This will cause a stall when the turn is long or very sharp. Reproduced multiple times in 2 flights and also another user confirmed that on FB. 

## Steps to Reproduce
1. Set Up CRUISE+ALTHOLD, POSHOLD and/or RTH on a switch
2. make the plane to do a full 180° turn in RTH or fly in 3D Cruise with a full stick turn or put plane into RTH
3. Stay ready to switch into Acro or Manual mode to catch the plane in case it stalls
4. Throttle will decrease massively in tight turns causing a stall 

## Expected behavior
Throttle should adapt to turns by throttling up to maintain airspeed and not throttling down. As it worked fine in 2.5 and previsous versions

## Suggested solution(s)
Fix Autothrottle behavior. 

## Additional context
Dump: 
[INAV_cli_AR_Pro_20201107_164330.txt](https://github.com/iNavFlight/inav/files/5505011/INAV_cli_AR_Pro_20201107_164330.txt)
[Flight1_log.zip](https://github.com/iNavFlight/inav/files/5505012/Flight1_log.zip)
[Flight2_log.zip](https://github.com/iNavFlight/inav/files/5505013/Flight2_log.zip)

Video with behavior: https://youtu.be/zI7pLo4DeDk
The logs are from the exact flights in the video so you can find the timestamp from the OSD. 

---

version
INAV/MATEKF722WPX 2.6.0 Nov  6 2020 / 14:17:31 (d0046f30)
GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

digitalentity, 2020-11-07:
Doesn't look like this is a bug. This behavior didn't change between 2.5 and 2.6.

From Flight2_log.txt:

At 3:20 log time plane enters a hard turn. At this point altitude error is +1.49m and plane pitches down slightly (~2 deg)
[image]

However, during the turn plane climbs more. The altitude error grows to 5.93m. Plane wants to dive more (20 deg at this point) which brings the throttle down to zero as configured:
file:///home/ksharlaimov/Pictures/Screenshot%20from%202020-11-07%2017-39-00.png

With the following parameters configured, the dive angle of 16.7deg would be enough to zero out throttle. And at max dive in automatic modes throttle will always be zeroed out.

```
set nav_fw_cruise_thr = 1350
set nav_fw_min_thr = 1050
set nav_fw_max_thr = 2000
set nav_fw_dive_angle = 20
set nav_fw_pitch2thr = 18
```

Possible improvements here:

1. We can use actual pitch value (not target pitch value) to calculate throttle correction.
2. As discussed in #6101, `TURN_ASSIST` (silently enabled in automatic modes) may pitch up too much  on wings, causing unwanted climb during a sharp turn. Merging that PR and finding a suitable pitch gain on wings might is probably a good idea for 2.6.

digitalentity, 2020-11-07:
@DzikuVx we should probably tweak presets to set `fw_turn_assist_yaw_gain` to 0.5 or lower on wings (per https://github.com/iNavFlight/inav/pull/6101#issuecomment-696600962 a value of 0.1 was good for Dart 250)

b14ckyy, 2020-11-07:
OK that makes sense. Maybe the AR Wing pro is just very sensitive too on pitch as it has a lot of reflex by default due to the Airfoil. I saw that pitch up on sharp auto navigation torns on multiple planes but that never caused an issue before. I think the AR Pro by design does not need a lot of pitch up in turns at all when trimmd well. In manual mode I may add 10% pitch up deflection on the stick in a 45° bank angle. I will play around with fw_turn_assist_yaw_gain tomorrow and try again. Not sure how well trottle control based on actual pitch will work especially in windy conditions or in case of a close stall situation when the nose drops and INAV want's to pull up. Might be problematic if it goes full pitch up but lowers the throttle due to down pitch.

digitalentity, 2020-11-07:
Looking at the log, it looks that the plane pitched up while the controlled commanded it to dive. This seems to be in line on observations made in  #6101). 

@b14ckyy, I made a typo there, the gain to reduce should be `fw_turn_assist_pitch_gain`. It was reported that 0.1-0.25 works good on wings that otherwise pitch up a lot during a turn in automated modes.

digitalentity, 2020-11-07:
@b14ckyy you can try zeroing out the `fw_turn_assist_pitch_gain` altogether, in the worst case it will cause plane to dive and lose altitude during a turn, in which case the altitude controller will increase the throttle and attempt to pull up which is less harmful than trying to climb with zero throttle.

b14ckyy, 2020-11-07:
Yeah that pitch up is what I see on all of my Delta Wings. Also on the 250G but they are all on 2.5.2. I will flash 2.6 Alfa2 again and set the pitch gain down to 0.2 for now and try from there. I will likely fly again tomorrow and report back then. But what confuses me still is: in the second flight the Cruise throttle was at 1350 (I lowered that in the first flight via adjustments). Duringt he loiter the throttle is down to 25% although the target altitude is pretty consistent and the plane looks pretty much level (beside some turbulences). Can this be caused by the pitch gain? In the video at 0:57

digitalentity, 2020-11-07:
`set nav_fw_pitch2thr = 18` is going to yield quite sharp response (about 20% throttle change on just 10deg correction).

I'm thinking about removing this gain alltogether and map max climb/dive angles to `nav_fw_min/max_thr`. This will make tuning easier as you'll set your min throttle to a value where your airplane dives at max_dive_angle without stalling. Similarly `nav_fw_max_thr` should be set to a value that allows plane to climb at `nav_fw_climb_angle`.

b14ckyy, 2020-11-07:
Hmmm don't like that idea. What happens when I use `nav_fw_allow_manual_thr_increase = ON` to fight against strong headwind? This migth then limit the throttle to my `nav_fw_max_thr = 1600` and I want to fly faster? Is it then also limited or can I then override that?
The advantaghe on the other had would be, that I can have smooth throttle control in climbs and have a more intense throttle cut on descents for better glide in navigation mode descents. 

But IMHO the better option would be to have a throttle up factor and a throttle down factor separately instead of P2T ratio combined for both. 

This wing as an example can glide at cruise speed of 70kph at a 7° nose down tilt. So no throttle needed at more pitch angle than that. But due to the fixed P2T ratio INAV will still push with 1220us throttle value instead of glide. 

Oh and the P2T on this plane is not tuned yet perfectly. But it is close.

b14ckyy, 2020-11-08:
@digitalentity 
Just went home from flight and what can I say. it works PERFECT! Buttery smooth turns in cruise and RTH. I have set `fw_turn_assist_pitch_gain=0.1` and that works perfect on the AR Pro. Even sudden and rapid turns in 3D cruise are absolutely smooth and no nose drop or any sign of stall. 

See: https://youtu.be/YpLyBQrFYjQ
At the beginning I make a turn in 3D cruise with full stick and at 2:15 it is a 180° turn with RTH. 

Now I just need to get rid of the horizon drift that is still pretty bad especially in POSHOLD. But that's another story.

b14ckyy, 2020-11-10:
I think we can close here. I'll post in the pull request that the default needs to be set lower to avoid this issue.

closed 2020-11-10 by b14ckyy