iNavFlight/inav #6273: Unlimited Inverted Hangtime Mode for multicopters with Blheli

mujtabachang, 2020-11-08:
## Current Behavior
When a multicopter goes inverted, there is this moment which gives us a brief time of hangtime due to inertia until the craft falls back towards the ground. Wouldn't be nice if we could get unlimited hangtime?

## Desired Behavior
It would be awesome if we could get a mode called Hangtime Mode. When the craft is inverted, we can flip the mode on, so that it can maintain the current altitude with motors reversing and thrust vector pointing towards the ground to maintain the current altitude.

## Suggested Solution
**Basically it's just an altitude hold mode with the difference being all the motors directions are reversed.** This is not 3D. 3D mode requires throttle at mid point. When this mode is activated, throttle is automatically maintained by the Flight controller. T**his mode can be set using momentary switch on the radio.** 
To use this mode, one must set motor direction to bidirectional in the blheli suite.
What if someone activates this mode while flying normally? We could put a check if the craft is inverted only then activate this mode. 

## Who does this impact? Who is this for?
Everyone who freestyles!

Let me know what you guys think.

closed 2023-03-06 by DzikuVx

mujtabachang, 2023-03-06:
Completed? Is this even implemented?

MrD-RC, 2023-03-06:
After 2 years, no one has looked at coding this at all. It is unlikely to ever be looked at. Also, how many people use 3D ESCs on their quads.