iNavFlight/inav #6276: Dangerous FW Launch mode bug in combination with reversible motors

FinalFrag, 2020-11-08:
I have setup where I use reversible motors on a fixed wing to have reverse thrust during landings. I have set up all the throttle values to be centered around a 1500ms "0-throttle" position instead of 1000ms. eg. nav_fw_min_thr = 1500, nav_fw_launch_idle_thr = 1500, failsafe_throttle = 1500

Full diff: https://pastebin.com/DQVad3NW

To use launch mode I do the following procedure:
1) Throttle stick at 1500ms position
2) Arm
3) Raise throttle slightly to activate launch mode
4) Jerk plane
5) Throw plane

Normally, when not using reversible motors, the motors would spin up at step 4.

When using reversible motors, motors go from 1000ms to 1500ms at step 3 (full reverse to stopped), stay stopped until step 4 and spin up as intended at step 4.

I think I have narrowed the issue down to the code in mixer.c:
```
int getThrottleIdleValue(void)
{
    if (!throttleIdleValue) {
        throttleIdleValue = motorConfig()->mincommand + (((motorConfig()->maxthrottle - motorConfig()->mincommand) / 100.0f) * motorConfig()->throttleIdle);
    }

    return throttleIdleValue;
}
```

I think all occurrences of mincommand should be replaced by the 3d neutral config item when the reversible motors feature is enabled. However, this method is used in a lot of places so I'm not comfortable to do the change myself and submit a pull request.

Another, maybe less correct (?) but safer fix is to change `static void applyFixedWingLaunchIdleLogic(void)` in navigation_fw_launch.c to not scale from getThrottleIdleValue() to navConfig()->fw.launch_idle_throttle, but to instantly go to that value.

FC Board name and vendor: Matek F405 Wing
INAV version string: INAV/MATEKF405SE 2.5.2 Aug  4 2020 / 10:24:57 (faaedc7cb)

digitalentity, 2020-11-09:
Indeed this is a bug, it's not trivial to fix though, so postponing to 2.7

FinalFrag, 2020-11-09:
> Another, maybe less correct (?) but safer fix is to change `static void applyFixedWingLaunchIdleLogic(void)` in navigation_fw_launch.c to not scale from getThrottleIdleValue() to navConfig()->fw.launch_idle_throttle, but to instantly go to that value.

Can we at least get this fix in 2.6?

digitalentity, 2020-11-10:
This affects a rather exotic setup - airplane with reversible motors, so I'd rather advise not to use launch mode with those for 2.6.0. We may look into adding a workaround for this in 2.6.1. A proper fix would requires considerable overhaul of the mixer and the whole throttle pipeline which is on the roadmap, but definitely not for 2.6

closed 2022-03-13 by DzikuVx