iNavFlight/inav #6277: Enable DJI : OSD_ESC_TEMPERATURE

Jetrell, 2020-11-09:
I would like to request the BLHeli_32 ESC temperature be enabled, so that it can be seen on the DJI OSD.
This would be helpful for fixed wing flight. Especially since the AIR UNIT displaces a lot of heat into the fuselage, that increases the ESC's operation temperature.

[image]
.

digitalentity, 2020-11-10:
DJI VTX hardware doesn't support this OSD item.

Jetrell, 2020-11-10:
@digitalentity Thank you for informing me of that. 
I had not personally tested it with Betaflight. I had just seen so people on Youtube that showed they had the ESC temperature symbol on the OSD in the DJI goggles. 
So I assumed it would work. I noticed it didn't appear on the OSD in iNav.  So now I know why.

I'll close this now.

Jetrell, 2020-11-10:
@digitalentity Can you please humor me with an explanation in the way MSP communicates differently with the DJI goggles in iNav compared to BetaFlight?
I just don't understand why  DJI supports ESC temperature on the OSD in BF. But can't for iNav?

Thanks

https://forum.dji.com/thread-203712-1-1.html

digitalentity, 2020-11-10:
Ok, so this was added recently. Reopening and will look into enabling ESC temp for INAV as well

Jetrell, 2020-11-14:
@digitalentity Thank you for adding this!

closed 2020-11-14 by Jetrell