iNavFlight/inav #6278: Toggle Modes

stuwe911, 2020-11-09:
## Current Behavior
For changing modes dedicated switches and channels have to be assigned. for me i often reach the limit of my RC soon. 

## Desired Behavior
Allow modes to be selected by Toggle switch since there can be only one mode active at one time.

## Suggested Solution
Modify iNAV modes so that they can be changed in a sequence. 
In the modes tab add a column with a check-box that inserts this mode into the toggle sequence.
Add a variable for delay until the selected mode becomes active.
Make mode in OSD blinking whilst selecting.

## Who does this impact? Who is this for?
Anyone that wants more modes available then he can spare channels and switches.

avsaase, 2020-11-10:
If you are using an OpenTX transmitter this is already possible. You can disconnect the trim switches from the sticks (you don't need them with INAV anyway) and use them as a momentary switch to increment and decrement a global variable that then offsets a channel. You need to use two logical switches to constrain the global variable to the desired range. If you setup the modes channel accordingly you get the desired functionality. The downside is that if you have multiple nav modes setup you sometimes have to cycle through some nav modes to get to the right one, which can cause some erratic behavior (for example switching from loiter to cruise to rth). I actually use this setup to cycle through a list of Adjustments and use another trim switch to change the values. 

If you do have enough switches but not the enough channels, you can use some logical switches and special functions to map them all to one channel.

stuwe911, 2020-11-10:
> If you are using an OpenTX transmitter this is already possible. You can disconnect the trim switches from the sticks (you don't need them with INAV anyway) and use them as a momentary switch to [...]

I admire your creativity - sounds like a pretty nifty hack.
I don't use OpenTX so unfortunately this will not help me. I would welcome a clean solution "per design". Also i would like to keep my RTH on the dedicated switch/channel and use a separate switch/channel for toggling through the less important modes during flight. thanks anyway

Beardyblair, 2020-11-22:
what transmitter do you use? maybe we could think of a solution.

stuwe911, 2020-11-22:
> 
> 
> what transmitter do you use? maybe we could think of a solution.

my trusty old Graupner MC-24 - but she works great apart from the limit of 12CH's
thanks for looking into this!