iNavFlight/inav #6280: Nav Launch Throttle Low Indication

breadoven, 2020-11-10:
## Current Behavior
Currently when using Nav Launch if you don't raise the throttle after Arming the motor won't start and the launch will fail. There is nothing to indicate that the throttle has not been set, just the NavLaunch beeps once armed which occur regardless of throttle position.

## Desired Behavior
Provide "beeper" feedback that the throttle is low after arming which changes to the normal NavLaunch beeper once the throttle has been set above low.

## Suggested Solution
Add additional beeper sound whilst the throttle is low after arming. The following simple change to `navigation_fw_launch.c` has been tested and does the job:

```
    if (!launchState.launchFinished) {
        if (calculateThrottleStatus(THROTTLE_STATUS_TYPE_RC) != THROTTLE_LOW) {
            beeper(BEEPER_LAUNCH_MODE_ENABLED);
        } else {
            LOW THROTTLE BEEPER SOUND ADDED HERE
        }
    }
```
The low throttle beeper needs to be "urgent" sounding and distinct from BEEPER_LAUNCH_MODE_ENABLED. Rapid short beeps such as BEEPER_ACTION_SUCCESS works well (bad choice of Beeper name admittedly but sounds right).

## Who does this impact? Who is this for?
Fixed wing flyers.

## Additional context
An alternative would be to have no NavLaunch beeper sound until throttle is set above low, indicating the Launch sequence hasn't been completed because the throttle hasn't been set correctly.
Other alternative solution would be to allow the motor to run with Throttle Low during NavLaunch. Only safety issue could be that the motor will start if the model is armed and then jolted whilst being carried to launch but then you shouldn't be arming it until you are ready to launch anyway so this shouldn't be an issue.

breadoven, 2020-11-10:
The following change in Flight/Mixer.c allows the motor to run during Launch with throttle low:

```
motorStatus_e getMotorStatus(void)
{
    if (failsafeRequiresMotorStop() || (!failsafeIsActive() && STATE(NAV_MOTOR_STOP_OR_IDLE))) {
        return MOTOR_STOPPED_AUTO;
    }
    if (calculateThrottleStatus(feature(FEATURE_REVERSIBLE_MOTORS) ? THROTTLE_STATUS_TYPE_COMMAND : THROTTLE_STATUS_TYPE_RC) == THROTTLE_LOW) {
        //if ((STATE(FIXED_WING_LEGACY) || !STATE(AIRMODE_ACTIVE)) && (!(navigationIsFlyingAutonomousMode() && navConfig()->general.flags.auto_overrides_motor_stop)) && (!failsafeIsActive())) {
        if ((STATE(FIXED_WING_LEGACY) || !STATE(AIRMODE_ACTIVE)) && (!(navigationIsFlyingAutonomousMode() && navConfig()->general.flags.auto_overrides_motor_stop)) && (!failsafeIsActive()) && (!isFixedWingLaunchDetected() || isFixedWingLaunchFinishedOrAborted())) {
            return MOTOR_STOPPED_USER;
        }
    }
    return MOTOR_RUNNING;
```
Code added:
**&& (!isFixedWingLaunchDetected() || isFixedWingLaunchFinishedOrAborted())**

Works on the ground, not flown though. Shouldn't be needed I guess if the warning beeper is enough of a reminder to set the throttle after arming and obviously the beeper reminder is the better solution to avoid problems with the motor stopping once Launch is finished.

dragnea, 2020-11-13:
Hey @breadoven . Have you seen this implementation? https://www.youtube.com/watch?v=Lh1rxddKeyU&ab_channel=DragneaMihai
[https://github.com/iNavFlight/inav/pull/5988](Pull request)
Maybe I will include the beep tone in that implementation too.
I'm using the new autolaunch since I've posted here. Dozens of flights so far and no issue. Just waiting for the admins to merge the code 👍 
Cheers!

MrD-RC, 2020-11-13:
That looks awesome Dragnea. The smoothing out transition at the end is sliky, I like it.

PR #6285 (2020-11-14): Autolaunch low throttle indication
This change adds a beeper sequence when launch mode is primed (enabled and armed), but the throttle is low.

Demo: https://youtu.be/axl-wPb-qqs

Fixes #6280
closed 2020-11-14 by digitalentity