iNavFlight/inav #6288: MSC broken on Matek F722-WPX

b14ckyy, 2020-11-14:
## Current Behavior
If MSC mode is enabled via CLI on the F722-WPX Flight Controller, the FC will freeze and not show mass storage device on PC

## Steps to Reproduce
1. Flash INAV 2.5.1 from Matek with SD and baro drivers or INAV 2.6 Alpha 2 
2. go to CLI and type "msc" and hit enter
3. Windows will play disconnect and connect sound and configurator shows "Rebooting" message
4. COM-Port still shown as available in configurator and device manager and no mass storage device will be shown

## Expected behavior
COM-Port should disappear and a mass storage device should pop up

## Suggested solution(s)
Bugfix

## Additional context
Tested on 2 different Windows 10 PCs. All drivers working for configuration and flashing
Tested with a F405 Wing FC and same INAV 2.6 Alpha2 Build and works fine there
Same issue reported by another user on Facebook with same FC and 2 different PCs
Possible also F722-PX affected and @MrD-RC will test that on his

---
- FC Board name and vendor: Mateksys F722-WPX Wing

- INAV version string: 
INAV/MATEKF722WPX 2.6.0 Nov  6 2020 / 14:17:31 (d0046f30)
GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

stronnag, 2020-11-14:
Duplicate of #5860

b14ckyy, 2020-11-14:
> Duplicate of #5860

Ah damn. I just recently mentioned that to @digitalentity and he said most likely a driver problem and I have excluded that now. I have not searched for the issue as he did not know about it apperently. So sorry for the duplicate.

closed 2020-11-14 by b14ckyy