iNavFlight/inav #6289: Rename AirMode

rsilk1949, 2020-11-15:
As a relatively new iNav user, I have been confused by the mode that is not a mode -- AIRMODE.
As I understand it, this setting affects several real modes at low to zero throttle.
So for the sake of clarity, I propose a change of name only in configurator to something like "Glide assist" or "Idle assist".

digitalentity, 2020-11-15:
Makes sense. `AIRMODE` is a historical name.

@DzikuVx @stronnag @shellixyz @fiam WDYT? I think we should somehow mark the modes that are not modes, but modifiers.

stronnag, 2020-11-15:
I really can't get excited about this one. We have a load of legacy stuff that is either confusing (waypoints that are not points, instead modifiers as well), as well as at least  one somewhat pejorative settings ... `moron_threshold` anyone.  I agree it's confusing, but so is a load of other stuff. Change it and get a flood of "where's AIRMODE gone".  We can't win either way.

DzikuVx, 2020-11-15:
I'm also not that excited. Airmode is not `Glide assist` nor `Idle assist`. The closest is "do not zero I-term on low throttle"

digitalentity, 2020-11-16:
Indeed, the closest description of AIRMODE from pilot's perspective is probably `ASSUME ALWAYS FLYING`, hence **AIR** mode.

closed 2020-11-16 by digitalentity

rsilk1949, 2020-11-16:
@stronnag I agree it is confusing, so how about "Low throttle support" or "Low throttle control" or  "Low throttle compensation"?

MrD-RC, 2020-12-05:
What about just losing the "Mode" part, as I think that's where confusion comes in?