iNavFlight/inav #6293: FixedWings: Loosing direction to home in strong winds

Pairan, 2020-11-15:
## Current Behavior
As fixed wings doesn't use a compass it tries to determine the flight direction by the (gps!?) movement. The direction you move will represent "FORWARD"! This is fine for most of the time unless the wind is stronger than your flight speed (in your desired direction)!

Today I endet up in very strong winds and got confused by my home point arrow was moving in several directions although I kept my wanted orientation and direction. I was considerable slow as I was fighting the wind ahead. So the wind moved me faster than my flight speed in my orientation. This turned out like described above and pointed my home direction to be all over the place.

Yet I managed to land on the last millivolts! 

## Desired Behavior
Pretty simple request: the home point should stay reliable and point to the position where the aircraft was armed initially

## Suggested Solution
puh ... dunno ... but you guys are good at the math stuff - I only drain batteries and make my videos ;)
Perhaps there is a way to detect that
1) the aircraft didn't rotate (left or right)
2) and so is drifting by the wind into another (unwanted) direction
?

## Who does this impact? Who is this for?
Affects everybody who is carried away by strong winds and need to find home via the home point (and I would think its in all iNav versions for fw aircrafts without compass). I flew the described situation in iNav v2.5.1

digitalentity, 2020-11-15:
That's complicated to do without a compass. GPS doesn't give you a sense of "forward", only "you're moving in that general direction". Workaround - increase cruise throttle to make sure the plain can fight the headwind

Pairan, 2020-11-15:
... so I did ... 11km/h with 90% Throttle on a twin motor S1100 with 4s battery at 12.8v :) 

[image]

So basically I should attach the Compass of the used BN-880 and be good for that sake?

FPVZaphod, 2020-11-15:
Check this out :-)

https://github.com/iNavFlight/inav/issues/6163

Same problem, with a possible solution...

0crap, 2020-11-16:
So wind or not, this is not as simple as get the current GPS position and somehow substract the home GPS location and now you know how the home arrow should point?

FPVZaphod, 2020-11-16:
No, because then it could only tell you towards which GPS coodinates you have to fly - but not in which direction they are. iNav only knows the flight direction by the change of the coordinates over time, and it has to assume a plane is always flying forward. But in high wind conditions sometimes it doesn's. 
That's why copters need a compass for navigation.

b14ckyy, 2020-11-17:
I have an idea for a possible solution.
Background:
pre 2.1 INAV always used only movement vector for heading calculation, even if compass enabled. In 2.1 that changed due to the issue that some GPS calculated bad headings and enabling the compass, then used compass heading for navigation. The problem now is:
Without compass the plane flies straight lines to the target point like home and compensates for crosswind, but with very strong headwinds, this can mess with the heading if the wind is faster than the cruise airspeed.
With compass the headwind is no problem and the minimum ground speed can do it's job to make it back home but crosswinds will cause the plane to massively drift sideways and flies a spiral back home.

So my idea:
Primarily INAV should continue to use the GPS heading for flight direction all the time as it gives the most precise heading and it compensates for crosswinds to always fly the shortest path.
But INAV should compare GPS heading with compass heading if compass is enabled. If the deviation between both directions is off by more than 45°, this would mean that the plane either fights against very strong crosswind, that is close to the airspeed, or the plane is already flying backwards due to strong headwind. In this case INAV should switch to pure compass navigation to find the correct heading. So it will turn into the right direction and then the min ground speed can kick in and do it's job.

If the difference between GPS and Compass is back within the 45° error range, it can go back to GPS heading navigation after a few seconds as the min groundspeed threshold will keep the plane on track and it can fly back home in a straight line.

So Compass should only be used for navigation if the GPS cannot determine a correct heading due to wind.