iNavFlight/inav #6296: 2.6.0 - Critical bug in launch mode

digitalentity, 2020-11-16:
## Current Behavior
When arming with launch mode switch toggled, the FC spins up the motors and freezes. It's not possible to disarm.

## Steps to Reproduce
1. Toggle the launch mode switch
2.  Arm

## Additional context
* This is built from the master branch today
* INAV/MATEKF405SE 2.6.0 Oct 28 2020 / 11:00:38 (ab051b18)
* CLI dump: https://pastebin.com/a3xuuCuJ

digitalentity, 2020-11-16:
Minimalistic diff to repro on OMNIBUSF4PRO

```
# version
# INAV/OMNIBUSF4PRO 2.6.0 Nov 16 2020 / 13:02:10 (8fe6ec54)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

mmix reset
smix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

smix 0 3 0 50 0 -1
smix 1 3 1 50 0 -1
smix 2 4 0 -50 0 -1
smix 3 4 1 50 0 -1

feature PWM_OUTPUT_ENABLE

map TAER

serial 0 64 115200 115200 0 115200

aux 0 0 0 1700 2100
aux 1 1 0 900 2100
aux 2 36 2 1700 2100

set acc_hardware = MPU6000
set acczero_x = 1
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set receiver_type = SERIAL
set serialrx_provider = SBUS
set platform_type = AIRPLANE
set model_preview_type = 8
set applied_defaults = 1
```

digitalentity, 2020-11-16:
Bisected to this commit 6d1a7f0f44bbbcb1b0ab1b55b3400beb2f11f874, part of #5988. 

@dragnea, I'm reverting #5988 - see #6299. Please have a look at what's broken there so we can re-merge it to 2.6 milestone.

digitalentity, 2020-11-16:
Looking at the commit message for 6d1a7f0f44bbbcb1b0ab1b55b3400beb2f11f874 (the failing commit) - there was a merge conflict in `navigation_fw_launch.c`. Probably it was resolved incorrectly and introduced a bug.

digitalentity, 2020-11-16:
Not that simple as it seemed. After the revert I still see the issue on the original setup. Minimalistic repro from https://github.com/iNavFlight/inav/issues/6296#issuecomment-727941774 doesn't trigger it anymore though

dragnea, 2020-11-16:
@digitalentity after the revert it worked for me. but i can't find what is the issue in the new code. nothing has been changed since i'm using it.

digitalentity, 2020-11-16:
@dragnea I've already root caused it to a different change - fix pending. The fact that it worked after reverting is probably related to compiler optimizing complex logic conditions differently.

dragnea, 2020-11-16:
I see it now. thanks for finding this

digitalentity, 2020-11-16:
Root cause:

`getMotorStatus()` is calling `navigationIsControllingThrottle()` to check if the Navigation core is in control of throttle command.

However, `navigationIsControllingThrottle()` is calling `getMotorStatus()` back to check if user is manually stopping the motors.

This creates an endless recursion and an instant crash when entering ANY navigation mode that controls throttle with a setting `nav_overrides_motor_stop = ALL_NAV`. This

#6305 fixes the issue and also makes `getMotorStatus()` more readable by eliminating a multi-layer logic expression.

PR #6305 (2020-11-16): [NAV/MIXER] Refactor getMotorStop() code for better readability
Due to a multi-level logic condition in `getMotorStop()` we missed the fact that in certain configuration `getMotorStop()` and `navigationIsControllingThrottle()` could call each other in an infinite loop causing a crash when entering a navigation mode that controls throttle.

Logic is now changed by creating a separate function `navigationInAutomaticThrottleMode()` to probe if our Navigation core is in mode that may be controlling throttle, provided mixer allows that. `navigationIsControllingThrottle()` now correctly evaluates if [... cut]
closed 2020-11-16 by digitalentity