iNavFlight/inav #6301: Safehomes should allow you to configure the distance threshold

tonyyng, 2020-11-16:
I love this idea of that new feature. however I don't agree with the 200m limit that is introduced.
On my permanent spot i have my offset distance at 300m and it works perfectly. 200m will only slightly improve the situation vs launch position as RTH but still crash the plane into the trees. By forcing me onto a 200m limit you're completely eliminating the whole offset landing idea for me to be used (and maybe others also).
Please use 200m as default but give us the option to change that limit at our own risk. I don't see why it cannot go up to at least 500m+
cheers

_Originally posted by @stuwe911 in https://github.com/iNavFlight/inav/issues/5995#issuecomment-727691714_

tonyyng, 2020-11-16:
Yeah that makes sense.  I'll make the change when I get a chance.  I can't make any predictions as to when this will be available in a release.  

As you suggest, I could add a configuration option that defaults to 200m and allows you to override it.  Do you think the override should have an upper limit?  I'm of the mind that if you are overriding, you should know what you are doing

The logic currently selects the first (enabled) safehome it finds that is nearer than 200m.  Perhaps with an increased configurable threshold, the logic could change to finding the nearest enabled safehome that is nearer than the threshold.

Let me know if you have any other thoughts.  Hopefully, others can comment with their suggestions as well.

stuwe911, 2020-11-16:
thanks very much for looking into this! I do agree with you 100% that you should only override if you understand what you're doing!

- i don't see any reason why there should be a limit at all. you can't forsee the usage of such a feature. i can't autoland at my spot. but i have a large field 1mile away where i could. this even allows me to move my start location and increase my range by auto-landing somewhere else. great new feature if without limits.

- add a new function that turns the entire SH feature to OFF-ON-AUTO
OFF = no offset function is used. RTH is where armed. default setting
ON = as existing without limits
AUTO = always pick the nearest SH if further then RTH (arming loctaion). Ignore any range limits set.
That would be super for the true long ranger. plan your emergency landing fields along your trip like you do in a real flight. 20miles out a problem occurs that enforces autolanding, it will go for the nearest SH that is active (max 8+?) and land there.
make this available in OSD menu for changing on the field.

I see situations where you want a flexible limit to be set. others where I want to configure many sites and always go to the nearest one. with this setup you'll leave all options for all cases open and it would be such a great new feature to pick your safe emergency landing spots on your trip! increases the safety of LR flying dramatically!!! go for it :)

Mateyhv, 2020-11-17:
@stuwe911, I believe what you are looking for is out of the scope of the safehomes which is bring the plane back in proximity to the pilot over a safe location.  I am of the opinion that SH should have a hard limit maybe 1000 to 2000m thats debatable but not more. People are not checking all parameters in each flight and without a limit you run the risk of a flyaway to a remote S of course running into obstacles, etc. If you actualy want to land in a remote location you should rely on a mission land waypoint. Iif an emergency landing is triggered I also doubt the plane will be able to navigate anymore. Its just going down in a more or less controlled way.
What you are looking for I believe is better oriented for improvements on the mission planner, alternative landing sites, etc.

MrD-RC, 2020-11-20:
I agree with @Mateyhv. I think a limit of 1Km should be plenty for a safe home feature. The safe home to me was a well defined safe home point, near to where you arm. Not for an alternate landing location. If should literally just be in the bigger field, next to where you launched, or a short distance away to avoid trees or other obstacles if emergency landing.

Having lots of landing spots far away, would be more for an emergency, alternate landing locations. This could be set up, but I think would be better as a more advanced feature. For example, you encounter stronger winds on a waypoint mission than expected (while out of radio contact). The system could estimate that you don't have enough power to return home, but there is an alternate landing site close to the route back, and would be reachable with your current power capacity. It could choose the closest alternate landing site to home, which is en-route. This way, it's less distance to retrieve, and there is more chance that video and control link can be recovered. Maybe have a system message in the OSD too, saying "En Route to Alt Landing Site 5" for example.

tonyyng, 2020-11-20:
I think I'll keep it simple and limit it to the following changes:
- A new configuration option. safehome_max_distance that is the maximum distance a safehome will be considered.  It will default to 200m and have an upper limit of 650m.
- When arming, the **_nearest_** enabled safehome within safehome_max_distance will be selected.  Previously, the **_first_** enabled safehome within 200m was selected.

The change to allow a configurable max distance addresses the OP's concern that his current offset distance of 300m wouldn't work.  The previous maximum for nav_rth_home_offset_distance was 650m.

I don't think we should support safehomes that are too far away for several reasons.
- The further away the safehome, the more likely you can't see the landing point and not know if there are people, cars or other obstructions at the time of your flight.
- This may not be obvious to everyone yet, since safehomes have not been released, but a safehome replaces home when arming.  So RTH and OSD information such as distance to home, and direction to home will point at the safehome, not where you armed.  The further away the safehome, the more confusing this difference becomes.
- Related to the last point, I don't know if the landing logic uses the arming point altitude when landing at the safehome.  The further away the safehome is from the arming point, the more likely the altitudes are different.

The original intent of this feature was for emergenies.  I lost radio control twice this year and my airplane circled down around my arming point, where there were other people, cars, sports field bleacher seating and a building.  My aircraft is sub 250g and did no damage, but it could have easily landed on/into the building or in a tree.

After implementing the logic, I defined safehomes for the handful of fields I fly at and didn't touch it again.  I'm not convinced there is a use case for needing to configure the behaviour at the field through the OSD menu.

stronnag, 2020-11-20:
@tonyng seems like an excellent solution in all respects.

stuwe911, 2020-11-22:
@tonyyng 
thank you very much for resolving the 200m offset limit that was the only concern. well solved and i fully support all your changes.

on the safe home landings i see the point why they are not suited to serve that purpose. i liked and support the idea to enhance MP with a similar function.

i still think that if we could program alternate landing sites with all gps/alt/hdg? infos it requires upfront it would only increase the safety. depending on your issue you're going down no matter what where it doesn't help if you see the people or not. that is always a risk you take when you takeoff with any plane. but if i.e. there is a town to the left and farming land to the right - it would help to define the right side as preferred landing area. define alternates and have FC go to "nearest" is standard in aviation.  many things can happen up there. in some cases it would've saved the plane in others not. but if there is a chance to increase the safety then I fully support any enhancement in that area.

thanks for your work Tony!

closed 2020-12-05 by stronnag

kasatka60, 2021-02-10:
Can increase the upper limit up to 1-2 km? The nearest open field is 1 km from the launch site.

tonyyng, 2021-02-10:
If the plane is going to land 1-2 km away, you don't know if it is safe to do so.  You should try to find some space at the field you are flying to land.  You could try to reduce nav_fw_loiter_radius.  This controls the size of the circles that the plane makes as it descends.  I use 50m, but depending on the size of your aircraft, it could be even smaller.  If there isn't large enough landing area at the place you fly, maybe you shouldn't be flying there.

kasatka60, 2021-02-11:
A safe landing field 1 km from the launch site. 
I just need to increase the limit, not the default setting of 200 meters.

tonyyng, 2021-02-11:
I'd suggest you create a new issue, since this one is closed.  No one (including me) will be reviewing this thread.

When creating a new issue, I'd suggest you include your launch GPS coordinates and explain why the nearest landing area is 1-2km away.

MrD-RC, 2021-02-11:
> A safe landing field 1 km from the launch site.
> I just need to increase the limit, not the default setting of 200 meters.
That is not the idea of safehomes. The safe home should be visible from where you are flying to ensure it is safe to land. Safehomes is not for remote landing sites.

rts18, 2021-02-12:
> 
> 
> Can increase the upper limit up to 1-2 km? The nearest open field is 1 km from the launch site.

@kasatka60 This feature has to be safe for the majority of users. @tonyyng @MrD-RC have brought up many good reasons why it should be near.

You can use `rest home position`. Found under the mode tab. 
Fly to your chosen landing location and flick the mode switch to enable this mode function. That will become your new home location.

PR #6333 (2020-12-05): Add safehome_max_distance option.  Choose nearest safehome.
Fixes #6301 

A new option ```safehome_max_distance``` (default 20000cm, max 65000cm) instead of the previously hardcoded 200m.

The logic also changed to select the nearest safehome, instead of the first enabled safehome within the maximum distance.

It's winter here, so I won't have a chance to flight test this change.  Maybe someone further south could give it a try.