iNavFlight/inav #6303: Strange TIM_USE_FW_MOTOR assignment in DALRCF405 target

Tygrys-1, 2020-11-16:
file DALRCF405/target.c

This look strange, as the S1 and S6 are using the same timer for FW_MOTOR and FW_SERVO. 
S6 should be FW_MOTOR

    DEF_TIM(TIM3, CH3, PB0,  TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_MOTOR, 0, 0),   // S1 (1,7)  <----------
    DEF_TIM(TIM8, CH1, PC6,  TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO, 0, 0),   // S2 (2,2)
    DEF_TIM(TIM1, CH3, PA10, TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO, 0, 0),   // S3 (2,6)
    DEF_TIM(TIM1, CH1, PA8,  TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO, 0, 1),   // S4 (2,1)  (2.3 2.6)
    DEF_TIM(TIM8, CH3, PC8,  TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO, 0, 0),   // S5 (2,4)  (2.2)
    DEF_TIM(TIM3, CH4, PB1,  TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO, 0, 0),   // S6 (1,2)  <-----------

digitalentity, 2020-11-17:
Indeed, this would create a problem if anybody is using S6. Although, a workaround is to stick to conventional PWM

Tygrys-1, 2020-11-17:
Does iNav has always two motors as a convention, or it is needed for something? This target has only one motor, so maybe it should be FW_MOTOR at S6?

digitalentity, 2020-11-17:
Two motors is not a convention. I'm not sure if any body is using this board for twin-motor or if S6 is in use at all. Most people fly either a wing (1 motor + 2 servos) or a conventional airplane (1 motor + 4 servos).

Tygrys-1, 2020-11-17:
Thank you. I have a small drone FC board with 4 outputs and a plane with 1 motor and 3 servos. But at current stage target is mapped as 2 motors and 2 servos. Can I remap (my private build) one motor to a servo in target.c, if both motors are on different timers now?
    DEF_TIM(TIM8,  CH2, PB8, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_MOTOR,    0), // PWM1 - PB8
    DEF_TIM(TIM4,  CH4, PB9, TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,                       0), // PWM2 - PB9
Can I change PWM2 to TIM_USE_FW_SERVO ?
(Excuse me for asking this here)

PR #6696 (2021-03-19): Fix the timer and DMA conflicts in the target, resolves #6303
Update target.c file for DALRCF405. Fix the timer and DMA conflicts in the target, resolves #6303
commit 7fa82f0a (2021-03-19): Update target.c

Fix timer and DMA conflicts in the target, resolves #6303
commit abce8455 (2021-03-19): Merge pull request #6696 from Tygrys-1/Tygrys-1-DALRC-Target-fix

Fix the timer and DMA conflicts in the target, resolves #6303
closed 2021-03-10 by Tygrys-1