iNavFlight/inav #6310: Force Baro only for altitude reading

Xedos9er, 2020-11-17:
## Current Behavior
GPS+BARO fusion is used for altitude estimation leading to sluggish reading and sometimes odd behavior when activating pos hold. Excample: do some acrobatic manouvers and switch to PosHold -> quad losses altitude and starts slowly climbing to desired altitude. You can see that altitude reading is very sluggish and needs some time to correct.
When using baro only on BF the altitude tracking is very fast without any significant delay and also very accurate.

## Desired Behavior
Enable cli setting for force BARO only like on BF

## Suggested Solution
add CLI inav_use_BARO_no_GPS

## Who does this impact? Who is this for?
faster and more accurate altitude reading. Better altitude tracking in Nav modes.

Mateyhv, 2020-11-17:
There is a specific weight for altitude estimation for baro and gps. I believe you can achieve what you want by bumping the baro to 100% and gps to 0%. Have not looked into it deeply but you may give it a try.

Martijn0001, 2021-09-29:
@Xedos9er  have you been able to test this suggestion? I have been thinking about this as well, once I noticed a similar behaviour.