iNavFlight/inav #6312: Auto launch is applying throttle immediately after arming in 2.6-RC2

avsaase, 2020-11-18:
## Current Behavior
Today I noticed that with 2.6-RC2, auto launch permanently enabled (and some idle throttle configured), the motor starts spinning directly when you arm the board, instead of after raising the throttle stick. This is potentially dangerous if you don't expect this and is a significant change from previous behavior.

## Steps to Reproduce

1. Have auto launch permanently enabled.
2. Not sure if this is required to reproduce, but I have some idle throttle configured.
2. Arm
3. Motor starts spinning
4. You still need to raise the throttle stick, otherwise the launch will not be detected.

## Expected behavior
Don't spin motor until throttle stick is raised.

## Suggested solution(s)
Could it be a problem with recent changes to auto launch?

## Additional context
[diff all](https://pastebin.com/yCZ5f2Ub)

---

- FC Board name and vendor: MATEKF411SE
- INAV version string: INAV/MATEKF411SE 2.6.0 Nov 17 2020 / 10:46:27 (1a490f4d)

digitalentity, 2020-11-18:
I noticed this as well. This is NOT how launch mode is designed to operate and needs fixing.

avsaase, 2020-11-18:
This seems similar to the bug you found in RC1, but other than the motor spinning up the flight controller performed normally.

digitalentity, 2020-11-18:
@dragnea, as you recently touched the launch mode code, can you please look into fixing this? If you're busy - I can have a look later this week.

dragnea, 2020-11-18:
@avsaase I did not intend to change the old behaviour. It is tested on my main planes, but not using permanent autolaunch. So I have to see what happens. Do you had a chance to test with autolaunch on a stick? At least we will know more. Here is 
 with some idle throttle: https://www.youtube.com/watch?v=Lh1rxddKeyU&ab_channel=DragneaMihai
@digitalentity I will have a look tonight or tomorrow after work. Thanks

b14ckyy, 2020-11-18:
Found  the reason: 

nav_overrides_motor_stop causes this. If it is OFF > all good. If it is set to AUTO_ONLY or ALL_NAV, motor spins at IDLE throttle when armed.

dragnea, 2020-11-18:
@b14ckyy yep. I'm working on another approapch. I don't know if it's ok@@

b14ckyy, 2020-11-18:
before the RC phase nav_overrides_motor_stop worked differently with just ON and OFF. And that seemed to work with your code. Maybe it's easier to fix there? @digitalentity

dragnea, 2020-11-18:
@b14ckyy Yep. I'm working on it. The actual implementation is using `getThrottleIdleValue()`. What if we use the min_command value, for the autolaunch. On the wings we don't need the idle throttle, right?

b14ckyy, 2020-11-18:
@dragnea launch idle throttle is absolutely important for a lot of planes! If you talk about motor idle, I'd prefer to keep it. I like to keep props spinning on planes with props that are clear from the ground. Esspecially scale planes.

dragnea, 2020-11-18:
@b14ckyy that's another thing. I'm talking about the autolaunch idle. it's only about of the autolaunch sequence. When you arm the plane, you want the motors to be off. Using getThrottleIdleValue() now will get the motors running. By usgin the min_command value, the autolaunch will don't care about the idle settings. but just in autolaunch.

dragnea, 2020-11-18:
@digitalentity  what do you think? https://github.com/dragnea/inav/commit/b672f5335c119014152fc7f4195854905b1d020e

b14ckyy, 2020-11-18:
OK I don't get it. The `nav_launch_idle_thr` is important. We can't remove that. 

If I arm now, the motor will spin at `throttle_idle` and NOT on `nav_launch_idle_thr`. When I then raise the throttle stick, the motor speed goes to `nav_launch_idle_thr`, that's correct. 

so `throttle_idle` must not be used when plane is armed and AUTOLAUNCH is active. But that's what is happening right now. 
If `nav_overrides_motor_stop` is set to OFF then it works fine. so here the new `nav_overrides_motor_stop` settings cause the motor to spin at idle as soon as nav launch is active.

dragnea, 2020-11-18:
@b14ckyy of course. I don't intend to remove nav_launch_idle_thr or anything from the autolaunch
`throttle_idle` it's used on the quads. At least I don't have something in my mind to be useful for planes

b14ckyy, 2020-11-18:
idle throttle can also be used on heavy planesin dives to act liek a airbrake as a spinning prop causes more drag than a still standing prop. Better check the code for `nav_overrides_motor_stop`. This should ignore LAUNCH mode as a navigation mode. Then it should work.

dragnea, 2020-11-18:
Yes, or to ignore the launch mode. Good catch with the nav_overrides_motor_stop. We can assume that the old or new autolaunch has nothing to do with this issue. I have tested right now in many situations. 
@avsaase please try with `set nav_overrides_motor_stop = OFF`
But still @b14ckyy I like the ideea to keep the motor off after arming in autolaunch mode, even if you want some `throttle_idle`. I mean that you arm->motor stays off(not idle)->raise the throttle->motor goes to launch the idle value->... and so on. I usually arm my wings while are on the ground to check the goggles, so if you want `nav_overrides_motor_stop = AUTO_ONLY or ALL_NAV` it's not possible to do that

b14ckyy, 2020-11-18:
Exactly 

> I mean that you arm->motor stays off(not idle)->raise the throttle->motor goes to launch the idle value->... and so on

This is how it was before and should be again. If `motor_stop=ON` the motor will not spin when you arm. And it should never do. So the bug must be in the `nav_overrides_motor_stop` now as this had no influence into launch mode before.

digitalentity, 2020-11-18:
Ok, so `nav_overrides_motor_stop` is the catch here. Odd, as mixer still considers `NAV_MOTOR_STOP_OR_IDLE` correctly.

Based on this snippet we shouldn't even get to `nav_overrides_motor_stop` checkL

```
    if (!failsafeIsActive() && STATE(NAV_MOTOR_STOP_OR_IDLE)) {
        return MOTOR_STOPPED_AUTO;
    }
```

digitalentity, 2020-11-18:
The `NAV_MOTOR_STOP_OR_IDLE` state flag is set by `applyThrottleIdle()`. The refactored launch code is missing a call to`applyThrottleIdle()` from `fwLaunchState_FW_LAUNCH_STATE_WAIT_THROTTLE()`:
https://github.com/iNavFlight/inav/blob/master/src/main/navigation/navigation_fw_launch.c#L246-L276

avsaase, 2020-11-18:
Yes, I'd like to keep `nav_overrides_motor_stop = ALL_NAV`, because I find it very unintuitive to be able to stop the motor when the autopilot should be controlling the throttle. I think the easiest it to disable `nav_overrides_motor_stop` when auto launch is active.

digitalentity, 2020-11-18:
Previous (pre-refactor) implementation of launch mode was setting the flag all the time before throw is detected:
https://github.com/iNavFlight/inav/blob/2.5.2/src/main/navigation/navigation_fw_launch.c#L186-L225

I guess adding the wait for non-zero throttle caused this bug.

digitalentity, 2020-11-18:
Should be trivial to fix. In progress.

dragnea, 2020-11-18:
> https://github.com/iNavFlight/inav/blob/master/src/main/navigation/navigation_fw_launch.c#L246-L276

@digitalentity  I've already done this. The motors still at the idle value until you don't set nav_overrides_motor_stop = OFF

dragnea, 2020-11-18:
But what do you think if we use min_command for keeping the motors low until we raise the throttle? In this case, @avsaase can use the option ALL_NAV

dragnea, 2020-11-18:
> Should be trivial to fix. In progress.

@digitalentity 
this, to use throttle_idle after arm: https://github.com/iNavFlight/inav/pull/6313/commits/b78ab3066f74a9cf99183d8128e474236b4cba99
or force to motor off after arm: https://github.com/iNavFlight/inav/pull/6313/commits/b672f5335c119014152fc7f4195854905b1d020e

breadoven, 2020-11-18:
Isn't the problem that NAV_CTL_LAUNCH is included in the following meaning the motor will run during launch with the throttle stick low if launch idle throttle is set ? 

```
bool navigationInAutomaticThrottleMode(void)
{
    navigationFSMStateFlags_t stateFlags = navGetCurrentStateFlags();
    return (stateFlags & (NAV_CTL_ALT | NAV_CTL_EMERG | NAV_CTL_LAUNCH | NAV_CTL_LAND));
```

dragnea, 2020-11-18:
Yes. This is better. So we have to remove NAV_CTL_LAUNCH

breadoven, 2020-11-18:
Well given Launch isn't supposed to work with the throttle low it doesn't make sense to include it in the Nav modes that allow the motor to continue running at throttle low.

Funnily enough I was messing about with a change to allow Launch to work with the throttle low throughout. Just Arm and throw and Launch only fully finishes once you raise the throttle, defaults to cruise throttle if the other end launch parameters are triggered (Time, altitude, stick waggle). Including NAV_CTL_LAUNCH as above might be handy for that. I had it working on the ground with the old code but now it's all changed so needs redoing.

dragnea, 2020-11-18:
The new code does the same thing as the old one, but it's separated in steps, so I can add easier the transition to another flight mode at the end of the launch sequence

digitalentity, 2020-11-19:
> Isn't the problem that NAV_CTL_LAUNCH is included in the following meaning the motor will run during launch with the throttle stick low if launch idle throttle is set ?

This is WAI. Launch is a navigation mode. Navigation is in charge of throttle. It's just navigation is NOT commanding motor stop when it should.

breadoven, 2020-11-19:
One thing that is apparent if `nav_overrides_motor_stop `is set to ALL is it's not possible to stop the motor by lowering the throttle stick once Launch is detected and the motor is running at Launch throttle. You can only stop it by disarming or waggling the sticks and lowering the throttle (with some delay, but delay fixed now perhaps ?). In the case of a failed launch the easiest and most intuitive way to stop the motor before the prop gets shredded hitting the ground is by lowering the throttle but this doesn't work with ALL selected. (works with AUTO and OFF).

And I couldn't replicate this problem. Motor doesn't start when Arming using various combinations of settings, just works the way it used to when ALL is set. Admittedly this is using a custom build updated 1 day ago from the iNavFlight Master but with no changes that obviously affect the Launch or Mixer code.

digitalentity, 2020-11-20:
@breadoven try branch. Launch mode is special and handles low throttle stick separately. It should behave the same way as it was in 2.5 now.

PR #6319 (2020-11-20): [LAUNCH] Fix a bug with throttling up in launch mode before throttle is raised
Fixes #6312, also some code style changes.
@dragnea, PTAL, this is what I ended up yesterday. Tested, working as intended.
closed 2020-11-20 by digitalentity