iNavFlight/inav #6314: 2.6.0-RC2 no control during launch "finishing"

shellixyz, 2020-11-18:
## Current Behavior

No control during launch "finishing"

See demo, no control from second 10 to 13:
https://www.youtube.com/watch?v=Ia97rpahRNU

## Desired Behavior

Control should be given back to the user as soon as the sticks are wiggled

avsaase, 2020-11-18:
I noticed this too today. On my first launch I also noticed another problem: the airplane pitched up 90 degrees when auto launch was finishing, but I had to wait until it completely finished before I could correct. I was not able to reproduce this on following launches so I can't completely rule out pilot error, though.

digitalentity, 2020-11-19:
Confirmed. Investigating.

dragnea, 2020-11-19:
@shellixyz I added a fix for this

closed 2020-11-19 by digitalentity