iNavFlight/inav #6315: Ad safetycheck to nav_rth_home_altitude

avsaase, 2020-11-18:
## Current Behavior
When the user erroneously sets `nav_rth_home_altitude` above `nav_rth_altitude`, the aircraft will climb when it reaches the home position.

## Desired Behavior
I think there is no real-world application for having the aircraft climb when it reaches the home position, so this should not be allowed.

## Suggested Solution
When `nav_rth_home_altitude` > `nav_rth_altitude`, ignore `nav_rth_home_altitude`. A check in the configurator would also make sense.

breadoven, 2020-11-18:
I think there could be a few limited cases where this could occur but you'd have to make a big mistake with the settings to end up with a significant climb at the home position. A warning in configurator might make sense. If the climb rate is too high and the plane is prop hanging then try changing `nav_auto_climb_rate `which is set at 5 m/s by default, too much for both climbing and descending on a fixed wing other than something small perhaps. This setting is only used for the altitude change to the RTH Home Altitude.

digitalentity, 2020-11-19:
Problem here is that `nav_rth_altitude` may mean different things depending on other settings. We can use `MIN(run_time_nav_rth_altitude, nav_rth_home_altitude)` during the RTH hovering above home procedure though.

Mateyhv, 2020-12-21:
I agree with breadoven. There would be a few cases where it might be useful to climb over the home position. Even in the case of pilots error, in regard of ground obstacles climbing over home position is in fact safer. As been mentioned a configurator warning would be good to point out a possible planning issue but I wouldn't put a hard restriction on it.