iNavFlight/inav #6316: Inav Led Ring behavior is not correct for Rover platform.

axelisart, 2020-11-18:
Hi,
When using Ring Led feature, INav consider throttle Zero as the minimum rotation speed for the Led Ring, which is correct for quad or plane but is not for Rover with reversible ESC. In this case the minimum speed is at center throttle (neutral at 1500). The result is that the behavior of led ring does not match speed of the rover. When the rover is stopper the Ring continue to turn at medium speed and the ring stop to turn at full rear speed (corresponding to throttle zero).
Thanks

MrD-RC, 2020-11-20:
Has anyone tested this on aircraft with reversible ESCs? I thinking this could effect everyone, not just rovers.

axelisart, 2020-12-11:
Hi, seems to work fine on Inav 2.6.0. Thanks soooooooo much

closed 2020-12-11 by axelisart