iNavFlight/inav #6317: Flight mode sensor never receives any data

Batperson, 2020-11-19:
I am running iNav2.5.2 with a Crossfire Nano rx / Radiomaster TX16S tx. I have enabled telemetry and I see all sensors except FM are receiving values. The telemetry screen on my radio never shows an * next to FM, indicating that data is not being received. I would like to resolve this problem so I can use iNav lua telemetry, could somebody help me to troubleshoot? Thanks.

[image]

Xedos9er, 2020-11-20:
Downgrade to opentx 2.3.9, it´s a but introduced by 2.3.10.

OptimusTi, 2020-11-20:
https://github.com/opentx/opentx/issues/8089

Batperson, 2020-11-20:
Thanks Optimus, looks like TBS have documented it here: https://www.team-blacksheep.com/tbs-tracer-manual.pdf
And they have fixed it in the nightly builds, so hopefully a fix will be released soon.

closed 2020-11-20 by Batperson

Pounii25, 2021-01-06:
> Downgrade to opentx 2.3.9, it´s a but introduced by 2.3.10.

I did't use 2.3.9... there is any way to download that .bin file? I really can't understand why companion force to the last one when you still able to flash previous version from your own back up. this function does not increase the safety and stability of the radio, it only causes problems