iNavFlight/inav #6324: DJI OSD - Display Air Speed via dji_use_name_for_messages re-purposing

Jetrell, 2020-11-20:
Currently @digitalentity has implemented a re-purposing of the `dji_use_name_for_messages`.  So that other warnings and data can be displayed on the OSD. #6287
This gives even more flexibility to the OSD in the DJI goggles.
The current layout is -

    // :W T S E D
    //  | | | | Distance Trip
    //  | | | Efficiency mA/KM
    //  | | S 3dSpeed
    //  | Throttle
    //  Warnings

I would like to request that a CLI command be added for the 'S' data. So that `Speed` can display as either `3DSpeed `or `Air Speed`.
This would be great for those of us that have air speed sensors hardware left over on our planes, from use with the analog OSD.

Thanks

mf0o, 2021-01-06:
I dont have an airspeed sensor but will look into this. Just got a contact who could test it properly

Jetrell, 2021-02-04:
> 
> 
> I dont have an airspeed sensor but will look into this. Just got a contact who could test it properly

@mf0o Sorry for the late reply. Thank you for looking into it :+1: 
I could test it for you. The target would be for an F722-WPX

Jetrell, 2021-02-16:
#6610

closed 2021-03-08 by Jetrell