iNavFlight/inav #6325: Climb Mode

FPVZaphod, 2020-11-20:
## Current Behavior
n.A.

## Desired Behavior
For longer Flights, sometimes it would be nice to enable a gentle and efficient climb as a flight mode. Usable together with Cruise Mode as an alternative for Cruise + AltHold. 

## Suggested Solution
An new Mode, "Climb Mode". Using two new variables, one for climb angle, and the other one for climb throttle. Probably a 3rd one,  "max autoclimb alt", where the mode switches to level flight to prevent unlimited climb. 

## Who does this impact? Who is this for?
Flixed Wing Pilots

## Additional context
Currently I do this in OpenTX, a Switch overriding Throttle to a certain level, overriding Flight Mode to Angle Mode, and Elevator Stick to slightly up.  With a flight mode this would be easier.

avsaase, 2020-11-20:
Not what you asked for, but since you are already programming in OpenTX you could use cruise mode instead of angle and put a few percent up elevator on a switch. That way INAV manages the throttle using the existing framework. If you have telemetry you can also program a max altitude after which the airplane levels off.

digitalentity, 2020-11-20:
I have a mix on my radio to add in a slider to my pitch channel. When I want gentle climb or gentle descent, I flip a switch to enable that mix and adjust the slider to get desired behavior.

FPVZaphod, 2020-11-22:
Ok, I will do in the Radio :-) Just need to use fixed values, there are no sliders on my Tango2.

MrD-RC, 2020-11-23:
@FPVZaphod you could use the E and F momentary switches to change a Gvar, which is added to elevator. Simplest way is to use that Gvar as the offset. Maybe limit the Gvar to +-20, and play a beep when it's at 0.

DzikuVx, 2023-10-27:
#9294

closed 2023-10-27 by DzikuVx