iNavFlight/inav #6328: Add version number to OSD

MrD-RC, 2020-11-22:
## Current Behavior
Currently, this does not exist..

## Desired Behavior
Select **Version** from a list in the OSD tab. Position it where you want it. The firmware will show it's version number in the OSD, where you position it.

## Suggested Solution
Add an OSD item to the firmware, and implement it in the configurator

## Who does this impact? Who is this for?
My initial thoughts were that this would be a useful too for content creators. Having the version number visible would make it easier to keep track of the version used when making the specific video. 

However, then I went on to believe it could also benefit tuners. If you have a specific tuning OSD layout, adding the version number here would also make it easier to keep track. 

Finally, I thought it may be beneficial to new iNav pilots, as most use the configurator version number when talking about problems. This would make the firmware version number more prominent. However, I also have an idea for the configurator that may help this last point too.

breadoven, 2020-11-22:
Could just be added to the Arming screen, avoid taking up room on the flight display.

MrD-RC, 2020-11-22:
That’s an option. But also, it would only take up room on the display if you allow it to. 

Just having it on the arming screen would mean you can’t show it along with your PIDs. Also, when I envisioned it, I actually wanted it there for the whole flight, so that it is visible in wherever you edit the video. 

I agree that it would be a good addition to the arming stats. Nobody really ever sees the boot screen. But I would still like to see it as an optional OSD element also.

Beardyblair, 2020-11-22:
Its already on the arming screen.

breadoven, 2020-11-22:
Just Position (Lat, Lon and Plus Code) date and time on mine. No version info.

Beardyblair, 2020-11-22:
You are correct - its engrained on my brain from the startup sequence

PR #6344 (2020-12-15): Add version to osd
Added an option to add the firmware version number to the OSD. It will also display the version number on the **ARMED** screen.

Fixes #6328 

[image]

[image]

PR for configurator element https://github.com/iNavFlight/inav-configurator/pull/1119
closed 2020-12-15 by DzikuVx