iNavFlight/inav #6330: Stop motors on low throttle

capblack, 2020-11-22:
Stop motors on low throttle with INAV 2.6.0 RC3 on my quad copter not work
Holybro Kakute F7 AIO 1.5V

Jetrell, 2020-11-24:
@DzikuVx I'm wondering if this issue is about the same thing I have noticed. 
Motor Stop does appear to work differently with 2.6, than before. 

I used to have Motor Stop turned ON with my larger multirotor and also **did not** have air mode permanently enabled. Although I did have `air mode` selected for one of the modes.
My reason for doing it this way, was so that I could pull the throttle down in `Horizon mode` and the motors would stop spinning, without having to disarm to them a on a Touch 'N' GO landing. This prevent the multirotor from becoming a lawn mower.
Although if I landed in `air mode`... the motors would still keep spinning, as expected.
Now Motor Stop will prevent the motors from spinning in all modes with the throttle down.. even `air mode`.

Is this part of a change to 2.6?

DzikuVx, 2020-11-25:
Looks like I will have to check. I do not recall any chnages to MR airmode handling thought

DzikuVx, 2020-11-25:
@Jetrell @Jetrell which airmode handling type are you using? `mc_airmode_type`

DzikuVx, 2020-11-25:
Tested and Airmode and Motor_stop are working correctly on Multirotors. 
Airmode becomes active only after throttle crosses the threshold (1300us by default). As long as throttle do not cross the threshold, motors are allowed to stop on Motor_stop. After crossing, motors will not stop as long as airmode is active.
disabling airmode requires throttle to cross `1300` again!

Jetrell, 2020-11-25:
> Airmode becomes active only after throttle crosses the threshold (1300us by default). As long as throttle do not cross the threshold, motors are allowed to stop on Motor_stop. After crossing, motors [...]

Thanks for looking into this. 
I was using `THROTTLE_THRESHOLD`. The `Threshold` point and the ones mentioned above, plus a few other settings, not related to this got messed up in the Diff upload to the FC. Notice after looking with WinMerge.
Reloaded and been out flying it, and its all fine. Sorry for the false alarm.

DzikuVx, 2020-11-30:
Since we have no new activity, I'm closing this one. Thanks for the participation!

closed 2020-11-30 by DzikuVx