iNavFlight/inav #6331: Unable to connect to ESC's through BLHeliSuite32

ArmyKnifey, 2020-11-22:
## Current Behavior
When attempting to connect to BLHeliSuite_32 via FC passthrough, error message 'Initialization of Flight Controller denied! Please check all motors are stopped!'. This issue occurred in both 2.6 RC2 & RC3. I flashed iNave back to 2.5.2 and was able to successfully connect to the ESC's through BLHeliSuite. 

## Steps to Reproduce

1. Connect battery to FC
2. Connect USB Cable to PC
3. Open BLHeliSuite32.exe 
4. Verify selected COM port is correct and press 'Connect'

## Expected behavior
BLHeliSuite should connect and allow me to read and program each ESC.

## Suggested solution(s)
Given BLHeliSuite works in iNav 2.5.2, something must have changed between releases. 

## Additional context

https://pastebin.com/i212ivsE

---

- FC Board name and vendor: INAV/MATEKF722SE 2.6.0 Nov 20 2020 / 23:58:58 (27266e34)
- INAV version string: GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

ArmyKnifey, 2020-11-22:
[image]

Jetrell, 2020-11-22:
@ArmyKnifey Just to confirm. I gave 2.6 RC3 a go with BLheli32 32.7.1.9. I have not had an issue with getting it to connect to a 4 in 1 ESC.
My FC is an F722 and the ESC is a Mamba F40
The Port should be displayed as -  STM virtual COMport when the FC is powered by a battery.

digitalentity, 2020-11-22:
@4712 maybe you can give a hint what might be causing it?

4712, 2020-11-22:
@ArmyKnifey I tried to reproduce, but could not, even not with pasting the [cli dump](https://pastebin.com/i212ivsE).
Please try to check what the log in BLHeliSuite32 shows.

[image]

Enable Log (Options... Enable log), then try a "Connect".
At log window: Log... Save To File... and save as log.tlg and post here as zip file.

ArmyKnifey, 2020-11-22:
@4712 Thanks for checking, I will enable the log as suggested and re-test today.

ArmyKnifey, 2020-11-23:
@4712 I have tried again today and encountered the same problem. Log file has been captured and attached as requested. 
[BLHeli_Log.zip](https://github.com/iNavFlight/inav/files/5580986/BLHeli_Log.zip)

4712, 2020-11-23:
@ArmyKnifey Thanks! Now I was able to reproduce.

[image]

@digitalentity While min_command = 1050 and ESC Protocol = Standard (PWM), it makes sense to have zero throttle at 1050.
If ESC Protocol = DSHOT then zero throttle is always at "1000", min_command should be ignored. MSP_MOTOR should reflect this (1000 1000 0 0 0 0 0 0). At least Betaflight handles it this way.

@ArmyKnifey Did you use the old stored settings after flashing 2.6.x?
Try in cli: `set min_command = 1000` and `save`

ArmyKnifey, 2020-11-24:
@4712 It's possible I did retain that value from a previous build. I did as suggested and set min_command = 1000 and everything now works. Thanks very much for your assistance!

digitalentity, 2020-11-25:
Indeed, it looks like a config data was retained from a previous version. `min_command` defaults to 1000, it indeed makes no sense to make it higher. Closing this issue. Thanks @4712 for your help!

closed 2020-11-25 by digitalentity