iNavFlight/inav #6336: New Waypoint type: "Circle/Loiter up/down"

tedk84, 2020-11-22:
## Current Behavior
At the moment, INAV fixed-wing will ascent/descent in a linear matter. 

## Desired Behavior
A new type would fit in like: the fixed-wing ascents/descents around the coordinates until it has reached the desired height.

## Suggested Solution
A solution might have one or two inputs: desired altitude and angle/feet per minute. A first approach would be to use the limits configured for the plane. 

## Who does this impact? Who is this for?
Waypoint users with airspace boundaries.

## Additional context
Context here: I have a area where I can legally climb up to XX meters. That area is around 200x200 meters. My airplane could circle up without having to create a lot of waypoints or to manually circle the plane.

stronnag, 2020-11-22:
As far as waypoint definition is concerned, you don't even need a new WP type. Just use the unused parameter 3 as a flag for this behaviour on relevant WP  types (WAYPOINT, POSHOLD_TIME).

breadoven, 2020-11-22:
Circular loiter climb/descend is already used to adjust the altitude to the RTH Home Altitude so the same method could be used for this. It uses an altitude climb/descent rate in m/s.

MrD-RC, 2020-11-23:
Wouldn't it be more efficient to use the method that GA autopilot uses? Have a target altitude at the waypoint, with target and maximum climb/descent rates. The goal being to be at the altitude when the waypoint is reached, but not start the climb/descent until necessary. If you're still too high/low perhaps employ the loiter climb/descent. In general, a spiral climb is going to be much less energy efficient that a climb en-route.

kasatka60, 2020-11-25:
Is it possible to implement waypoints landing on fixed wing? How on ardupilot?

closed 2022-02-15 by breadoven